greatscottgadgets/hackrf #868: hackrf release bundles don't automate installing pre-built firmware

rfc6919, 2021-04-06:
Downstream packagers of the hackrf tools don't install the `firmware-bin` pre-built firmware from release bundles, because (in [at least one case](https://github.com/Homebrew/homebrew-core/issues/74398)) there's no supplied automated way to do so. It'd be nice if it was obvious to downstream packagers that the pre-built firmware should be made available on users' systems.

Ideally this shouldn't be part of the firmware build process, since packagers shouldn't have to also have a working cross toolchain and the opencm3 &c deps installed. I'm not sure the nicest way to do this, perhaps even just including a note in the Readme.md that packagers should also install `firmware-bin` to `/usr/share/hackrf/firmware-bin` would help?

### Steps to reproduce
1. download release bundle
2. look for a `make install` option to install pre-built firmware

### Expected behaviour

There's a way to install the included pre-built firmware other than just copying the `firmware-bin` directory

### Actual behaviour

There isn't a way to install the pre-built firmware other than just copying  the `firmware-bin` directory

ZeroChaos-, 2021-04-06:
This is what I do for gentoo, it could be easily scripted if desired

https://github.com/gentoo/gentoo/blob/master/net-wireless/hackrf-tools/hackrf-tools-2021.03.1.ebuild#L34-L43

mossmann, 2022-09-27:
We feel that copying firmware binaries to a system directory is a distribution-specific solution that should be up to packagers, so we've decided not to provide a script for that. Thank you for showing how it can be done, @ZeroChaos- !

closed 2022-09-27 by mossmann