greatscottgadgets/hackrf #869: Exception on exit for very fast (<1s) single-shot sweeps

dmaltsiniotis, 2021-04-06:
### Steps to reproduce
1. Compile master branch from source.
2. Run: `hackrf_sweep.exe -f 100:1000 -w 100000 -l 16 -g 24 -1 -r out.bin`

### Expected behaviour
hackrf_sweep should run a single-shot and exit without errors.

### Actual behaviour
An exception occurs when hackrf_sweep is existing/cleaning up, the exception is listed in the screenshot below.

### Version information
**Operating system**: Windows 10 Pro

**hackrf_info output:** 
[image]

### Output
```
C:\dev\hackrf_tools>hackrf_sweep.exe -f 100:1000 -w 100000 -l 16 -g 24 -1 -r out.bin
call hackrf_sample_rate_set(20.000 MHz)
call hackrf_baseband_filter_bandwidth_set(15.000 MHz)
Sweeping from 100 MHz to 1000 MHz
Stop with Ctrl-C

Exiting...
```
_When the exception is acknowledged the output continues normally._
```
Total sweeps: 1 in 0.12459 seconds (-107374176.00 sweeps/second)
hackrf_close() done
hackrf_exit() done
fclose() done
exit
```

I believe this issue is arising from this commit: https://github.com/mossmann/hackrf/commit/5361b3a7f43cce012dbd3525aec6ddfdb6c756ad. The code that sets sweep_count is encapsulated in an if statement that first checks to see if 1 second has passed before processing some output stats. However, if hackrf_is_streaming() returns false before 1 second has passed, control is moved toward the cleanup/stats code where sweep_count is printed out again at the end but this time is, unfortunately, uninitialized. 

I believe a simple fix for this use-case would be initializing sweep_count to 0 on declaration, which will result in a _slightly_ inaccurate exit message of `(0 sweeps/second)` but upside is application will not crash.

dmaltsiniotis, 2021-04-06:
Created PR #870 to address, if the proposed solution is acceptable.

dmaltsiniotis, 2021-04-23:
For tracking: PR #870 was folded into #874 which also addresses this issue as well as actually gives a more accurate projected sweep counter per second for sub-second sweeps.

mossmann, 2021-04-23:
fixed in #874
Thank you for tracking this down!

closed 2021-04-23 by mossmann