greatscottgadgets/hackrf #871: hackrf_sweep: stdout binary format seems to be corrupted (file out still okay)

dmaltsiniotis, 2021-04-06:
### Steps to reproduce
1. Run the following two commands and observe the output files
2. hackrf_sweep.exe -f 100:1000 -B -w 100000 -l 16 -g 24 -1 -r out1.bin
3. hackrf_sweep.exe -f 100:1000 -B -w 100000 -l 16 -g 24 -1 > out2.bin

### Expected behavior
I expected that the out2.bin file should look _similar_ in structure and format to the out1.bin file insofar as the lengths of each binary record.

### Actual behavior
In the out2.bin file that was piped from stdout, there seem to be the occasional extra byte polluting the power value integers in the frequency bin portion of the binary data. I came across this error because a 3rd party tool I was using (QSpectrumAnalyzer) was failing to parse the binary data from the latest version of hackrf_sweep.

When specifying an actual file to output to, the binary data appears correct and consistent. When reading from standard input, the binary data is corrupted.

### Version information
**Operating system**: Windows 10 Pro

**hackrf_info output:**

### Output
Console (stderr in the case) output is the same for each command
```
C:\dev\hackrf_tools>C:/dev/hackrf_tools/hackrf_sweep.exe -f 100:1000 -B -w 100000 -l 16 -g 24 -1 -r out1.bin
call hackrf_sample_rate_set(20.000 MHz)
call hackrf_baseband_filter_bandwidth_set(15.000 MHz)
Sweeping from 100 MHz to 1000 MHz
Stop with Ctrl-C

Exiting...
Total sweeps: 1 in 0.12549 seconds (0.00 sweeps/second)
hackrf_close() done
hackrf_exit() done
exit
```

```
C:\dev\hackrf_tools>C:/dev/hackrf_tools/hackrf_sweep.exe -f 100:1000 -B -w 100000 -l 16 -g 24 -1 > out2.bin
call hackrf_sample_rate_set(20.000 MHz)
call hackrf_baseband_filter_bandwidth_set(15.000 MHz)
Sweeping from 100 MHz to 1000 MHz
Stop with Ctrl-C

Exiting...
Total sweeps: 1 in 0.18038 seconds (0.00 sweeps/second)
hackrf_close() done
hackrf_exit() done
fclose() done
exit
```

In my command line example `-f 100:1000 -B -w 100000 -l 16 -g 24 -1`, the expected binary record lengths should be 224 bytes. 4 bytes at the start of each record specifying a record length of 220, 16 bytes (two 8 byte values) after that, and 204 bytes after that representing the floating point power values for each bin. I discerned this from QSpectrumAnalyzer's code, as I couldn't find a document explaining the binary format.

If you open the bin1 and bin2 files in a hex editor, and set the column width to 224 bytes, it becomes visually apparent there that extra bytes are being added to the piped version of the file:

out1.bin (the correct looking data):
[image]

out2.bin (the "bad" looking data):
[image]

I'm currently trying to track down exactly where/why the occasional extra byte is being sent to stdout.

I fully admit that I may have a major misunderstanding of the binary format of hackrf_sweep or a misconception what is actually going on here. This is my attempt at documenting what I _think_ is a bug while trying to use the most recent version of QSpectrumAnalyzer and hackrf_sweep.

dmaltsiniotis, 2021-04-07:
I believe I have tracked down the issue and submitted a fix under PR #872. Of course, the underlying issue turned out to be a bit of a rabbit hole, as it usually goes.

Try as I might, I just could not find any obvious problems with the hackrf_sweep code sending a spurious '\n' to the 'outfile' variable anywhere. This lead me back to analyzing the binary data more in depth, and looking at each byte sequence of `0x0A 0x0D`. Something that was nagging me from my earlier analysis was that I had assumed `0x0A 0x0D` was being _added_ to the byte stream, but in every case of the data record lengths being shifted, it was only off by ONE byte, not TWO bytes... This bothered me for some time until now, coming back to the issue it occurred to me that either 0x0A, or 0x0D may _already exist in the byte stream_ and only a single byte is being added.

A few cursory searches showed both 0x0A and 0x0D do exist independently and somewhat randomly (as expected) in the correct file output. So only one of the bytes is being _added_... this was the information that clued to me into the fact that this could be a carriage return auto-format issue specifically related to Windows pipes when using ```>```.

Some creative googling lead me to a somewhat dated, but very relevant article here: https://stackoverflow.com/questions/18294650/win32-changing-to-binary-mode-childs-stdout-pipe

This, it turns out, is the real issue. I think the appropriate fix in this case is the following: If compiling in a windows environment, and the binary output flag -B is set, then make a call to set the stdout pipe mode to binary as well, and don't let the standard c library mangle the data by assuming you forgot to include a '\r' with your '\n', which really wasn't a '\n' to begin with.

Cheers,

-DM

mossmann, 2021-04-23:
fixed in #872 

Thanks for figuring this out!

closed 2021-04-23 by mossmann