greatscottgadgets/hackrf #875: Using the hackrf on a conda environnement

NicolasCaseau, 2021-04-09:
Hello, 

I'm trying to use the hackrf on a conda environnement, I installed gr-osmosdr but I still need the Library libhackrf to be able to use the card. Do you know if I can install this Library using conda ? Or could you tell me if there is another way ? 

Thanks and have a good day :) 

Nicolas

PureRockets, 2021-04-30:
I am in the same situation. My conda install on Windows is not recognizing the HackRF. When I install on Windows from the binaries, it recognizes it. But I need to run from a conda environment.

straithe, 2021-10-29:
Sorry I haven't responded to this issue in a timely manner. I'm not familiar with Conda, so I'm going to try to follow along with your experience and get HackRF to work in this environment. Can you send me your steps to install Conda, what OS you tried to install Conda on, and any other relevant details?

denis-sp, 2021-11-12:
straithe, you can install anaconda from the official site https://www.anaconda.com/products/individual
As far as I understand there are guys on conda-forge channel who maintain gnuradio and some additional blocks, including osmosdr. Unfortunately it is built without hackrf support. But with support of soapy. I believe to support hackrf it's necessary to make recipes for libhackrf and soapy hackrf module. All dependencies for libhackrf are already on conda-forge. I've tried to build it and succied, even thouth few patches for cmake files are needed. I believe all dependencies to build soapy module are also there (except libhackrf). But I have no experience with soapy though and not sure about it.

straithe, 2021-11-12:
What conda-forge channel are you talking about, @denis-sp? I don't have reference for where this channel would be. I'd really like to check it out. I also don't have any experience with soapy either, so I'll get to learn lots of things trying to work this out. Thank you!

denis-sp, 2021-11-12:
Anaconda has a channel system. Channels are kind of repositories. You can install packages from there. Conda-forge is the biggest repository with unofficial packages. https://anaconda.org/conda-forge

denis-sp, 2021-11-12:
To build libhackrf in conda environment it was necessary to change following files:

FindUSB1.cmake

  FIND_PATH(LIBUSB_INCLUDE_DIR libusb.h
    PATHS ${PC_LIBUSB_INCLUDEDIR} ${PC_LIBUSB_INCLUDE_DIRS}
    PATH_SUFFIXES libusb-1.0)
I've added PATH_SUFFIXES

FindFFTW.cmake

IF (FFTW_LIBRARIES)
include_directories(${FFTW_INCLUDES})
find_library (FFTW_LIBRARIES NAMES ${FFTW_LIBRARIES})
ELSE(WIN32)
find_library (FFTW_LIBRARIES NAMES fftw3f)
ENDIF(WIN32)

IF ELSE block apparently was buggy and else never executed.

I believe for recipe it can be necessary to change installation path. To fit conda directory structure. I haven't tried it yet though.

denis-sp, 2021-11-14:
@straithe https://github.com/denis-sp/conda-hackrf-recipe Here is a recipe I made.. I'm not sure if everything as it should be, but it seems working for me. It's for Windows only though.

straithe, 2021-11-14:
Thank you for the update! I won't have access to my Windows computer for a few weeks. The soonest I will be able to try this out is early December. For now I am going to tag this as help wanted so members from the community might see and step in to help out.

denis-sp, 2021-11-14:
@straithe I've changed a bit previous recipe and made a recipe for hackrf soapy module https://github.com/denis-sp/conda-soapysdr-module-hackrf It seems working. But for some reason I have narrow band there.
[image]
.

denis-sp, 2021-11-15:
There is a bandwidth option there. I haven't try anything real yet, but I believe it works fine.

straithe, 2021-11-16:
@NicolasCaseau, I believe the recipe provided by @denis-sp solves your issue. Can you please try this solution and confirm?

marcusmueller, 2022-01-02:
Related: https://ham.stackexchange.com/questions/20359/the-transmission-and-reception-block-for-the-hackrf-does-not-work-windows-11/20365#20365

I'd honestly love to see both the `hackrf` as well as the `soapysdr-module-hackrf` package of @denis-sp being extended to more than just windows (that's mostly copying bld.bat to build.sh, and replacing a few `%variables%` with `${variable}`, far as I can see), and to become part of `conda-forge` (which is mostly [a short PR against a single repo](https://conda-forge.org/#add_recipe)).

Doing that would make the hackRF accessible through the gr-soapy blocks that are available by default in modern conda-based installations of GNU Radio (and conda is by now probably the most popular way to get GNU Radio on Windows, but certainly not the only one; the GNU Radio wiki recommends it for Windows to get current versions, thanks to @ryanvolz' extensive work on conda packaging).

ryanvolz, 2022-01-02:
I'm happy to help get these packages on conda-forge, and add Linux + macOS support. With the existing Windows recipes it should be straightforward. It's been on my to-do list for a while, but I don't personally have a hackrf to test with so I didn't want to try to support something without at least some willing testers. Seems like they've been here all along!

straithe, 2022-01-02:
I'd be happy to test things out on Linux (specifically Ubuntu), @ryanvolz! Let me know how I can help.

ryanvolz, 2022-01-14:
Hey all, I haven't forgotten about this. I've got a pull request in to add hackrf packages to conda-forge [here](https://github.com/conda-forge/staged-recipes/pull/17534), and I have the soapy module package almost ready to go once that one is merged. I'll ping this issue again when things are ready to get some testing.

straithe, 2022-01-14:
Thanks for the update!

ryanvolz, 2022-01-31:
The packages are now on conda-forge!

For testing, I'm partial to the [CondaInstall](https://wiki.gnuradio.org/index.php?title=CondaInstall) guide I wrote on GNU Radio's wiki, but the key part is to install a base conda distribution like [Miniforge](https://github.com/conda-forge/miniforge) and then you'll be able to install `hackrf` and `soapysdr-module-hackrf` from the conda-forge channel.

Let me know if you find any problems with the packages. I plan on adding them to my [Radioconda](https://github.com/ryanvolz/radioconda) installer soon, so it would be good to head off any problems before I do that.

straithe, 2022-01-31:
Thank you for the update! @NicolasCaseau would you be interested in helping test?

murat-sever, 2022-10-30:
Hi @straithe, I am using GNU Radio under conda environment on my Windows machine. After I have installed miniforge, I created an environment and installed GNU Radio and `soapysdr-module-hackrf` package into my environment as @ryanvolz's [conda install guide](https://wiki.gnuradio.org/index.php?title=CondaInstall). Then I was able to use my HackRF within GNU Radio Companion successfully.  
[image]

straithe, 2022-11-01:
That's fantastic to hear! Thank you for the review @murat-sever. And thank you for your work @ryanvolz. I'm closing this issue as completed. 🎉

closed 2022-11-01 by straithe