greatscottgadgets/hackrf #877: Not receiving FM with Hackrf One on GNUradio but can receive on SDR#?

shinobirain, 2021-04-16:
Hi Everyone!

I recently started using the hackrf one on my windows 10 pc and have having issues with the first tutorial from great scotts in that I cannot seem to receive/see any of the FM signals I see in SDR#. Here is the link to the tutorial I am using with the time stamped right has hes about to run the project.

Below are some pictures that I have attached to show what I have going on right now in the project file.

[image]
[image]
[image]
[image]

A few background things:

1. Currently running the 2021.03.1 Hackrf One Firmware and GNUradio 3.8

2. The command prompt does indicate warnings that say they are over written by something else that is very similarly titled (shown in the picture below). There are many more above; however, I'm not sure if that's effecting the output...?
[image]

I have also used SDR# with my Hackrf One and have been able to receive FM radio and listen to them as well.
[image]

Thank you for your help and suggestions in advance!

miek, 2021-04-19:
You have the gains set a bit higher in SDR#, so you could try increasing them in the osmocom source properties.
Also, try enabling some averaging in the QT GUI sink (~20 works well for me). That will make it easier to see the signals above the noise.

shinobirain, 2021-04-27:
Hey Miek, 

  I apologize for the delayed reply, I have tried what you suggested and am still receiving the same results as shown in the photos above.

straithe, 2021-05-31:
I suspect that there might be something wrong with gr-osmosdr. What is the output in the GNURadio Companion console when running the flow graph?

closed 2021-06-30 by straithe