greatscottgadgets/hackrf #880: HackRF One version??

josooik, 2021-04-23:
What difference does the HackRF One 2014 version and the HackRF One 2020 version make?
[image]
[image]

ellisdee110, 2021-11-07:
I can't find out what these chinese vendors mean either. Anyone know?? i assume either marketing or they couldn't get exact components anymore and used what was available. I ordered a "2020" to get a closer look.

straithe, 2021-11-08:
The HackRF hardware versions that have been manufactured by Great Scott Gadgets, and the changes between them, are listed here: https://hackrf.readthedocs.io/en/latest/list_of_hardware_revisions.html 

We do not have intimate knowledge of the hardware produced by other manufacturers and whether they line up with any of our hardware revisions. I'm closing this issue as I don't have any other knowledge to provide. You could try contacting your reseller for more information.

closed 2021-11-08 by straithe