greatscottgadgets/hackrf #881: No power on usb

michaeltak, 2021-04-23:
### Steps to reproduce
1.
2.
3.

### Expected behaviour
Tell us what you expect should happen

### Actual behaviour
Tell us what happens instead

### Version information
**Operating system**:

**hackrf_info output:**

If you are reporting a problem that involves third party software 
(GNU Radio, Gqrx, etc), please report the version here.

### Output
```
Insert any commandline or build output here
```

straithe, 2021-04-23:
Have you tried using a different cables?

michaeltak, 2021-04-25:
yes tried it all  and 2 things on the pcb are broken... don't know how... Just felt off have this board for ages now...

michaeltak, 2021-04-25:
[image]

miek, 2021-04-26:
If those parts are missing, that would certainly explain the problem.
Could you post some photos of your board?

michaeltak, 2021-04-27:
gonna make pics when home

michaeltak, 2021-04-28:
[image]

michaeltak, 2021-04-28:
[image]

michaeltak, 2021-05-03:
never mind... can't find the problem... so it's for the bin...

pickeditmate, 2021-05-03:
Looks like it's missing L10, and C146 and R52. Do you want a CSV of the parts and their values or do you have it already?

michaeltak, 2021-05-04:
this I gonna order I hope this is it 

https://www.mouser.be/ProductDetail/963-LBM2016T4R7JV
https://www.mouser.be/ProductDetail/739-TYA2016104R7M-10
https://www.mouser.be/ProductDetail/581-04024D106MAT2A
https://www.mouser.be/ProductDetail/810-YFF18PC0J475M
https://www.mouser.be/ProductDetail/71-CRCW0603-49.9-E3

straithe, 2021-05-05:
Please keep us updated!

michaeltak, 2021-05-06:
the site charge sending cost to high ... cant fint the righjt parts for sure so

closed 2021-06-13 by straithe