greatscottgadgets/hackrf #883: Freeze in Gqrx with hackRF 2021-03-1 driver version

Potomac, 2021-04-24:
### Steps to reproduce
1. Install Gqrx software, 2.14.4 version
2. Use hackRF (2014 version)
3. Update firmware to 2021-03 version
4. Use linux (archlinux)

### Expected behaviour
Gqrx should no randomly freeze during operation

### Actual behaviour
Randomly, when switching frequency mode (AM, FM, LSB etc...) then Gqrx can freeze, I hear no sound from hackRF, it can happen 30% of the time when switching frequency mode,

if I use 2018 hackrf driver version then there is no freeze, no bugs, all is Ok with Gqrx,
my hackRF is a 2014 PCB version, firmware has been updated to 2021-03

### Version information
archlinux 64 bits

hackrf_info version: 2021.03.1
libhackrf version: 2021.03.1 (0.6)
Board ID Number: 2 (HackRF One)
Firmware Version: 2021.03.1 (API:1.04)

Gqrx 2.14.4

Potomac, 2021-04-25:
If I use the official appimage of Gqrx (last version, same as archlinux package : 2.14.4) then there is no freeze, no bug with hackRF 2021-03-1 driver :
https://github.com/csete/gqrx/releases

Potomac, 2021-04-25:
But the appimage can also contain hackRF lib in 2018-01 version, that can explain why there is no freeze.

straithe, 2021-04-25:
Have you tried your HackRF with other software on Arch?

Potomac, 2021-04-25:
Yes, I tried 2 softwares : gnu radio and hackTV, both work with hackRF 2021-03-01.

Gqrx freezes randomly with hackRF 2021-03-01 when switching modes (AM, FM, LSB), and sometimes when I start/stop hackRF with the menu "file -> start dsp", and "File -> stop dsp", it seems that the driver in 2021-03 version may hang the device.

Potomac, 2021-04-25:
@straithe : do you have a test script/binary that can check if all features, functions of hackRF driver 2021-03-01 are Ok ?

I use 2 commands for checking transmission and reception and it seems ok : 
hackrf_transfer -t /dev/zero
hackrf_transfer -r /dev/null -s 21500000

But I feel these commands are not enough for testing hackRF driver, if Gqrx 2.14.4 manages to crash the 2021-03 driver (and not with the 2018 version) then it is worth to investigate, there is perhaps something wrong in the 2021-03 driver version about some functions, a regression.

miek, 2021-04-26:
I haven't had any luck reproducing this so far (running Ubuntu 20.04 & pybombs GR/GQRX).

Could you run gqrx in a terminal, reproduce the issue, and paste the terminal output here?

Potomac, 2021-04-26:
@miek : Don't forget that ubuntu 20.04 is not a rolling release distro, some libraries and linux kernel may not be at the last version, 
Archlinux is a rolling release (all packages are up to date), then bugs/regression can occur more easily,

I ran gqrx in a terminal, when the freeze occurs then the output doesn't show unusual information : 

```
$ gqrx
gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.8.2.0
built-in source types: file fcd rtl rtl_tcp uhd hackrf bladerf rfspace airspy soapy redpitaya 
gr::log :WARN: file_source0 - file size is not a multiple of item size
Resampling audio 96000 -> 48000
gr::fft: can't import wisdom from /home/cesar/.gr_fftw_wisdom
BandPlanFile is /home/cesar/.config/gqrx/bandplan.csv
BookmarksFile is /home/cesar/.config/gqrx/bookmarks.csv
[INFO] [UHD] linux; GNU C++ version 10.2.0; Boost_107500; UHD_4.0.0.0-0-unknown
[ERROR] SoapySDR::loadModule(/usr/lib/SoapySDR/modules0.7/libhackrfSupport.so)
  duplicate entry for hackrf (/usr/lib/SoapySDR/modules0.7/libHackRFSupport.so)
ATTENTION: default value of option vblank_mode overridden by environment.
gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.8.2.0
built-in source types: file fcd rtl rtl_tcp uhd hackrf bladerf rfspace airspy soapy redpitaya 
Using HackRF One with firmware 2021.03.1
```
perhaps there a debug mode for qgrx in order to have more output at the terminal ?

Potomac, 2021-04-28:
I made a git bisect and I found the faulty commit : 

```
1442014a80b5ce33f16ef665538c2881b37d3a68 is the first bad commit
commit 1442014a80b5ce33f16ef665538c2881b37d3a68
Author: Matioupi <mathieu.peyrega@gmail.com>
Date:   Wed Oct 16 12:03:08 2019 +0200

    Modified hackrf_stop_tx and hackrf_stop_rx to first join the transfer thread
    before setting the receiver to OFF mode (cf. issue #650)

 host/libhackrf/src/hackrf.c | 9 +++++----
 1 file changed, 5 insertions(+), 4 deletions(-)
```
https://github.com/mossmann/hackrf/commit/1442014a80b5ce33f16ef665538c2881b37d3a68

We should revert this faulty commit in order to fix the crash about stop_tx/tx commands that might occur on recent hackrf one chinese model, when using 2021-03-01 hackrf driver version.

The git bisect log : 

```
git bisect start
# good: [cc7f599677e222a19a2ba4336a23e1fb507cf9f6] Pin OS X builds to Xcode 7.3 for now
git bisect good cc7f599677e222a19a2ba4336a23e1fb507cf9f6
# bad: [e6eb4ba29bbe5dc2fcd092e394188bc10a8bad54] Merge pull request #848 from mossmann/pre-release
git bisect bad e6eb4ba29bbe5dc2fcd092e394188bc10a8bad54
# good: [56b3bd0bed544cf40c9287c6d2b1c811608eda57] Merge pull request #546 from dominicgs/operacake_gpio
git bisect good 56b3bd0bed544cf40c9287c6d2b1c811608eda57
# good: [1c7bf39bda81accb67d04c46aa08808d863180b8] Merge branch 'develop'
git bisect good 1c7bf39bda81accb67d04c46aa08808d863180b8
# bad: [92ee1447143aed8d6b7e291e98e4139190206aac] Merge pull request #699 from miek/ui_disable
git bisect bad 92ee1447143aed8d6b7e291e98e4139190206aac
# good: [923f9fe617c9dcf00c97c18f5aa79e51720b9e59] Merge remote-tracking branch 'mossmann/master'
git bisect good 923f9fe617c9dcf00c97c18f5aa79e51720b9e59
# bad: [058276d010d437486c4e0ee4e0b7fe3f3e52f836] Merge pull request #671 from yhetti/acrylic_case
git bisect bad 058276d010d437486c4e0ee4e0b7fe3f3e52f836
# good: [1569737109283ef944c3ae4d803bbcbb3a523fbf] Merge pull request #607 from dominicgs/portapack_ui_opera_cake_coexistence
git bisect good 1569737109283ef944c3ae4d803bbcbb3a523fbf
# bad: [8ff56c615fa8835e8a0db36ed754e2793ada9a9b] Merge pull request #661 from mgesteiro/hackrf_transfer-fix
git bisect bad 8ff56c615fa8835e8a0db36ed754e2793ada9a9b
# bad: [1442014a80b5ce33f16ef665538c2881b37d3a68] Modified hackrf_stop_tx and hackrf_stop_rx to first join the transfer thread before setting the receiver to OFF mode (cf. issue #650)
git bisect bad 1442014a80b5ce33f16ef665538c2881b37d3a68
# good: [208fae75389e1888c6118f12b79c4ac7522fe372] Merge pull request #645 from jboone/master
git bisect good 208fae75389e1888c6118f12b79c4ac7522fe372
# first bad commit: [1442014a80b5ce33f16ef665538c2881b37d3a68] Modified hackrf_stop_tx and hackrf_stop_rx to first join the transfer thread before setting the receiver to OFF mode (cf. issue #650)
```

Potomac, 2021-04-29:
Reverting this commit is not enough for solving freezes in Gqrx, as recent commits in hackrf change also start/stop rx, tx functions,
1442014a80 is just the first bad commit, the problem may be more deep, related to thread management, usb transfert, it will need more debugging for solving this bug.

ivan-kulikov-dev, 2021-05-04:
I have problem with hackRF one chinese model (bought in 2017 year) with hackrf-2021.03.1 firmware too.

straithe, 2021-11-03:
Thank you for the video. That helps! Can you give me a screen shot of your right hand menu before and after you change from WFM to AM, then to WFM? I am updating my setup and trying to follow along with your steps exactly and I can't quite see what buttons you are pressing there.

Potomac, 2021-11-04:
Hello,

here is a new screen capture, this time the freeze occurs after the change WFM to AM.

https://user-images.githubusercontent.com/1340993/140243083-50b7aa64-fc6b-4fa8-a058-647fef492068.mp4

straithe, 2021-11-15:
This issue may be related to #916. We are going to try to recreate that issue, address it, and hopefully provide a solution that you might be able to use as well. I'll update you as we have more information.

martinling, 2021-12-31:
Now that I've figured out how #916 happens, I suspect that this issue has the same cause.

Could you test with my fix in #1029 and see if that solves this issue for you?

Potomac, 2022-01-01:
@martinling : Can you provide a patch file compatible with the master branch of hackrf ?

Your current commit is not compatible with the master branch of hackrf (date of 2021-03), because I tried to apply your modifications to hackrf.c file and I get a lot of errors during compilation, about undeclared functions (HACKRF_OPERACAKE_MAX_BOARDS, and unknow types like hackrf_operacake_freq_range).

martinling, 2022-01-01:
@Potomac sure, here's a patch against 2021.03.01: [bug-916-rebase-2021.03.01.patch.txt](https://github.com/greatscottgadgets/hackrf/files/7797860/bug-916-rebase-2021.03.01.patch.txt)

You can also get the same result by cherry-picking my commit `b346790e`. It applies cleanly on top of the `v2021.03.01` tag.

Potomac, 2022-01-01:
@martinling : I tried your patch, I see no real difference, the bug is still here, gqrx freezes randomly after changing settings like frequency mode (AM, FM etc...), activating/deactivating RDS mode for example.

I suggested you to test with gqrx (not your test source code), and try to change settings like frequence mode, RDS, you have to test for a long period (sometimes the freeze doesn't occur immediately, you have to retry sometimes 20 times before the freeze occurs).

martinling, 2022-01-02:
I'm afraid I haven't been able to reproduce this problem here. I'm running gqrx 2.14.4 on Debian. With firmware and libhackrf from 2021.03.01, I can switch back and forth between FM and AM over 100 times without it hanging.

gozu42, 2022-01-02:
i am somewhat able to repro what is probably the same issue, even though i trigger it without any retuning.
fedora-35 vm, hackrf passed in through usbip.

gqrx starts up fine, i can rx some fm.
but if i repeatedly stop/start i am seeing ... 
- after a repeat start gqrx again will "receive" unchanging data (wiggly line not wiggling, unchanging pattern in waterfall)
- after a stop the rx led stays on
- the next attempt to start will lock up gqrx
- if i kill gqrx in this state, and start it again, it will lock up right away (rx led stays on)
- if i kill gqrx, then run something like hackrf_sweep -1, the sweep will work, the rx led goes dark, gqrx works again (for a while)

wireshark usb capture of a gqrx session during the issue. 
[gqrx-stall-usb.zip](https://github.com/greatscottgadgets/hackrf/files/7799215/gqrx-stall-usb.zip)
- 0sec start wireshark
- 9sec start gqrx
- 16.5sec start rx (working)
- 18.3sec stop rx
- 19.5sec start rx (not working)
- 29sec killed gqrx

martinling, 2022-01-02:
@Potomac could you perhaps try running the [test program](https://gist.github.com/AlexFWulff/7d6e75aaffb7215c2a175a2992db70e1) (hackrf_loop.c) from #916?

If this is a hackrf bug rather than a gqrx one, it seems like it must involve some failure during a quick stop & start of the hackrf - which is exactly what that test program exercises, over and over again.

You should be able to build it against your installed libhackrf with:
``gcc `pkg-config --cflags libhackrf` hackrf_loop.c `pkg-config --libs libhackrf` -o hackrf_loop``

Then if you run `./hackrf_loop` you should get output like the following, repeating indefinitely:

```
Start
Not started yet
Valid length: 262144
Valid length: 262144
Valid length: 262144
Valid length: 262144
Valid length: 262144
Valid length: 262144
Valid length: 262144
Valid length: 262144
Valid length: 262144
Valid length: 262144
Valid length: 262144
Valid length: 262144
Valid length: 262144
Valid length: 262144
Valid length: 262144
Finally started
```

If you see anything else (e.g. errors, hanging, or `Not started yet` repeating), then post it here - what you see may give us a clue as to what's going wrong in the gqrx scenario too.

martinling, 2022-01-02:
@gozu42, was that result with or without my fix from #1029?

gozu42, 2022-01-02:
> @gozu42, was that result with or without my fix from #1029?

```
libhackrf version: git-b346790e (0.6)
Firmware Version: git-a83bfc4b (API:1.05)
```
so libhackrf with the #1029 changes, firmware from m0/buffer branch which "shouldnt" matter for this.
(and according to ldd, gqrx is using the right libhackrf version)

martinling, 2022-01-02:
Aha! I've reproduced it now - using the same commits as @gozu42.

I wasn't able to get it to happen by switching AM/FM or the start/stop button, but using the keyboard shortcut (Ctrl-D) for start/stop and just holding it down, I triggered it after a few seconds. Terminal output shows:

```
Using HackRF One with firmware git-a83bfc4
Failed to start RX streaming (-1000)
Failed to stop RX streaming (-9999)
OOOOFailed to start RX streaming (-1000)
Failed to stop RX streaming (-9999)
Failed to start RX streaming (-1000)
Failed to stop RX streaming (-9999)
Failed to start RX streaming (-1000)
Failed to stop RX streaming (-9999)
Failed to start RX streaming (-1000)
Failed to stop RX streaming (-9999)
Failed to start RX streaming (-1000)
Failed to stop RX streaming (-9999)
```

I'll investigate further.

martinling, 2022-01-04:
I had to [patch](https://lists.osmocom.org/hyperkitty/list/osmocom-sdr@lists.osmocom.org/thread/OIEQODYWNF7INMXRU6NY2MWCRJS77THA/) gr-osmosdr to display the error messages to go with those error codes, but having done so, it looks like the same problem as #916: a `Resource busy` error from `hackrf_start_rx()`, probably indicating that some transfer was left in flight:
```
Failed to start RX streaming (-1000: Resource busy)
Failed to stop RX streaming (-9999: unspecified error)
Failed to start RX streaming (-1000: Resource busy)
Failed to stop RX streaming (-9999: unspecified error)
Failed to start RX streaming (-1000: Resource busy)
Failed to stop RX streaming (-9999: unspecified error)
```
So it seems like my patch in #1029 isn't a complete fix yet. I'll see if I can work out what the new failure mode is.

martinling, 2022-01-04:
It's taken me a while to actually catch this happening in gdb, but I've now done so.

At the point where `hackrf_start_rx()` is reporting `Resource busy`, the first of the four libusb bulk transfers attached to the device has a `status` of `LIBUSB_TRANSFER_COMPLETED`. The other three show `LIBUSB_TRANSFER_CANCELLED`. So far, this is normal. If that transfer was already completed when `cancel_transfers()` was called, but not restarted yet, then `libusb_cancel_transfer` would have returned an error for it, and not changed it to the `CANCELLED` state. It should be fine to restart that transfer.

But when `prepare_transfers` calls `libusb_submit_transfer` to restart the completed first transfer, it gets a result of `LIBUSB_ERROR_BUSY`. I don't know why - it seems to be caused by some internal state of libusb. Apart from the `status`, the transfer struct looks identical to that of the cancelled ones, and changing the `status` to `LIBUSB_TRANSFER_CANCELLED` doesn't allow `libusb_submit_transfer` to succeed.

Next step is to build libusb from source so I can step through `libusb_submit_transfer` and see what it doesn't like.

martinling, 2022-01-10:
@Potomac I think I've now fixed this with the new commit `38ecfa1` I've just added to PR #1029. Would you be able to test the new version and see if it solves the problem for you?

martinling, 2022-01-11:
Apologies, I pushed an incomplete version - the commit to test is https://github.com/greatscottgadgets/hackrf/pull/1029/commits/9182f1ca4cba6c3962f34dde7884248ed4747c96.

Potomac, 2022-01-11:
@martinling : thanks for your commit, I will test it soon as possible.

Potomac, 2022-01-11:
@martinling : Like the last time : can you provide a patch file compatible with the 2021-03 version of hackrf (last release version) ?
I don't master the git cherry-pick command.
Thanks.

martinling, 2022-01-11:
@Potomac sure, here you go - this should apply to the 2021.03.01 release and includes both fixes from #1029:
[bug-916-v2-rebase-2021.03.01.patch.txt](https://github.com/greatscottgadgets/hackrf/files/7850476/bug-916-v2-rebase-2021.03.01.patch.txt)

Potomac, 2022-01-12:
@martinling : I made the test of your patch with Qqrx, and I still have the bug, no change.

Then I tried the test program "hackrf_loop" and I see the message "not started yet" after few seconds :

Finally started
Start
Not started yet
Valid length: 262144
Valid length: 262144
Valid length: 262144
Valid length: 262144
Valid length: 262144
Valid length: 262144
Valid length: 262144
Valid length: 262144
Valid length: 262144
Valid length: 262144
Valid length: 262144
Valid length: 262144
Valid length: 262144
Valid length: 262144
Valid length: 262144
Finally started
Start
Not started yet
Not started yet
Not started yet
Not started yet
Not started yet
Not started yet
Not started yet
Not started yet
Not started yet
Not started yet
Not started yet
Not started yet

martinling, 2022-01-12:
Thanks for testing this @Potomac. In my case I'm no longer getting problems in gqrx, but I've seen that pattern happening in `hackrf_loop`, and @gozu42 noted it in #916 too.

I don't yet understand what's happening in the case where it just gets stuck at `Not started yet` without any errors being reported. I've poked at it with GDB and can't see what the problem is. The shutdown seems to have worked correctly, the restart seems to have successfully resubmitted the transfers, the transfer thread is looping and isn't hung, but the transfer callback never gets called.

I'm going to get the USB analyzer out next and check whether data is actually showing up on the wire in this case.

Potomac, 2022-01-12:
@martinling : what is sure is that the previous hackrf driver version (2018) doesn't have the bug,  hackrf_start_rx() and hackrf_stop_rx() run without problems with 2018 version on my hackRF.

So a solution would be to track the difference in source code between 2018 and 2021-03 version, I tried with a git-bisect command but the results were inconclusive, the first bad commit found by git is : 
https://github.com/greatscottgadgets/hackrf/commit/1442014a80b5ce33f16ef665538c2881b37d3a68

but reverting https://github.com/greatscottgadgets/hackrf/commit/1442014a80b5ce33f16ef665538c2881b37d3a68 is not possible, as commits made after need this faulty commit.

martinling, 2022-01-12:
Unfortunately because this involves a race condition, the problem can come and go as a result of completely unrelated changes if they affect the timing of the code execution. We can't rely on testing and bisecting - we need to understand what the actual issue is to make sure it's solved properly.

`1442014` made the stop functions kill the transfer thread, but that behaviour was removed later - the transfer thread now persists as long as the device is open, just as it did before that commit. So it's already effectively reverted anyway.

martinling, 2022-01-16:
I don't see a great way to attack this with more logging, because the symptom is that something _doesn't_ happen. As far as I've been able to tell so far, everything is being started up correctly but no data comes back.

Setting an environment variable of `LIBUSB_DEBUG=4` allows to see what's happening at the libusb level, but I see no further clues there. The attached log is from a Windows machine on which I can reproduce this quite consistently. The interesting part starts at line 2248. The control transfer to set the transceiver mode to RX is submitted, and completed successfully. Then the four bulk transfers are submitted, also successfully.

And then we wait for the transfer callbacks, which just never come. We can see that the transfer thread hasn't hung, because every 500ms it times out and starts `libusb_handle_events_timeout_completed` again.

It looks like there may be a device-side issue involved, so I'll be looking at that side of things next.

martinling, 2022-01-16:
The attached log is from Windows 10 running `hackrf_loop` against libhackrf with the #1029 fixes. The failure to start occurs at line 2248.
[failure-to-start.txt](https://github.com/greatscottgadgets/hackrf/files/7877992/failure-to-start.txt)

martinling, 2022-01-29:
I have confirmed that there is a firmware-side bug which can cause RX stop/start to appear to succeed, but stall after the first 16KB of data. I've started a separate issue for this in #1042.

Potomac, 2022-01-29:
Thanks Martinling for continuing the investigations,
you think it's a firmware bug, do you think that bug is present since the beginning of the project ? (from the first version of the firmware ? 2014 ?)

martinling, 2022-01-29:
I'm certain that it's a firmware bug, because I have captured the USB traffic directly, and we can see that the host has correctly told the device to stop & start, but then the device then stops sending data after the first 16KB. The host is continuing to ask for more data and the device doesn't send any more. That puts the blame clearly on the firmware.

At the moment I don't know how that bug happens on the firmware side, so I can't tell how far back it goes in the project history. And because it's intermittent and seems to be affected by timing, bisecting can't be relied on to find out. But it's interesting that you've have no problems with the 2018 version, so I'll look at the differences between there and 2021.

My next step will be to get a JTAG/SWD debugger hooked up to the HackRF's MCU and look at the firmware state when this happens. That should hopefully reveal something about the cause.

Let's keep further discussion about the firmware bug on #1042.

In terms of the original topic of this issue - the freeze in gqrx - my conclusion at the moment is that the same symptom can happen for three different causes. Two of those causes are host-side bugs that I believe to be fixed in #1029 (but that needs a little more work, plus review and merge). The third cause is the firmware bug which I'm now tracking in #1042.

martinling, 2022-02-03:
@Potomac I believe I've now fixed the firmware bug as well. If you're able to test #1029 now I think you should have no more freezes in gqrx. But you will need to update the firmware as well as the host side. Let me know if you need any help.

Potomac, 2022-02-03:
Can you provide a patch file compatible with 2021-03 hackrf version ?
The firmware file is built automatically when building hackrf ?

martinling, 2022-02-03:
Sure, here you go:

[bug-916-v3-rebase-2021.03.01.patch.txt](https://github.com/greatscottgadgets/hackrf/files/7998177/bug-916-v3-rebase-2021.03.01.patch.txt)

To build and flash the firmware you can follow the instructions [here](https://github.com/greatscottgadgets/hackrf/blob/master/firmware/README).

Potomac, 2022-02-03:
@martinling : I tried your patch and I flashed the firmware : the bug is still here, I can easily trigger the bug with Gqrx,

I tried also the hackrf_loop program : I get also the bug with the message :

Start
Not started yet
Not started yet

Perhaps you should test with a configuration similar to mine : 

- old PC (old CPU intel core2 quad Q9650 3 Ghz)
- a rolling release distro like archlinux, in order to be sure to use the last release version of libraries like libusb, and the last version of linux kernel

If I test with the appimage version of Gqrx (2.14.5) then there is no bug.

gozu42, 2022-02-03:
can you please post output from hackrf_info on your current configuration?
and perhaps doublecheck that both test program and gqrx are using the right libhackrf?
something like "ldd hackrf_loop | grep hack" and "ldd `which gqrx` | grep hack". 

i was able to repro the problem on both test program and gqrx before, but with the combination of the host+fw fixes the hangs are now gone here, on both high powered PC and low-powered raspi.

Potomac, 2022-02-03:
Well I think I made a mistake about the firmware file, I think I still use the 2021.3 version of firmware,

I didn't manage to build the new firmware file, the compilation fails, can you send me a compiled version of the new firmware file ?

gozu42, 2022-02-03:
> can you send me a compiled version of the new firmware file 

[hackrf-fw-7057235a.zip](https://github.com/greatscottgadgets/hackrf/files/7998952/hackrf-fw-7057235a.zip)

Potomac, 2022-02-03:
@martinling  @gozu42  : I finally manage to flash the firmware with the correct version send gozu42,

the bug seems gone when I use hackrf_loop program,
and also no bug when I use Gqrx,

Thank you very much Martin for this excellent job.

martinling, 2022-02-03:
That's fantastic news @Potomac, looks like we've finally solved this one. It's been quite an adventure!

gozu42, 2022-02-03:
the reported version there for the lib might actualy be an artifact of the take-release-and-manually-patch process.

my flow for the build was ... 
* going to the PR linked https://github.com/greatscottgadgets/hackrf/pull/1029
* at the top, find the "x want to merge to y from z", the z part https://github.com/martinling/hackrf/tree/bug-916
* `git clone -b bug-916 https://github.com/martinling/hackrf hackrf-martling`
* then build as usual

the result is that all versions in hackrf_info output directly give the git commit used:
```
hackrf_info version: git-7057235a
libhackrf version: git-7057235a (0.6)
Firmware Version: git-7057235a (API:1.05)
```

Potomac, 2022-02-04:
It would be interesting to have a nightly build channel for hackrf (like firefox), in order to download and test the current  beta/alpha version of hackRF binary (including the firmware).

Thanks again Martinling for the resolution of the bug, I hope these commits will be included soon in the master branch of hackrf.

martinling, 2022-02-04:
That's a good idea but I have some worries about it. We would need to provide quite a lot of different binaries - we have three supported boards for the firmware, and several supported operating systems for the host tools. Some users will always download the wrong thing without understanding, and that can create a lot of support headaches.

A simple change we could make is to retain the build products of our existing Github Actions builds, which [can be done](https://docs.github.com/en/actions/advanced-guides/storing-workflow-data-as-artifacts). We run these builds every time commits are pushed to master or a PR, so if we kept the results you could just look at the latest CI run for any branch, and download prebuilt binaries for libhackrf, the host tools and the firmware. But it would depend on the user knowing how to correctly install these in the right places I expect.

I'll discuss it with the team.

Thanks also to you @Potomac for all your work in reporting this problem, experimenting with things, and testing fixes over the last few months. I don't think we would have tracked down all the different bugs involved without your persistent efforts.

closed 2022-02-08 by mossmann