greatscottgadgets/hackrf #885: Added TCXO breakout board to hackrf one. Getting 0x41 from debug of register 0

wizarDD03, 2021-05-03:
I've recently tried to add a TCXO to the P22 hackrf header. After installing I ran this from a connected linux machine:
```bash
$ hackrf_debug --si5351c -n 0 -r 
[  0] -> 0x41
```

From the documentation I should expect to see a 0x01 (TCXO used) or 0x51 (TCXO not used) value for register 0.

What does 0x41 signify? 

I also did a diff of the registers with and without the TCXO and these are the differences:
With TCXO:
```bash
[  0] -> 0x41
[222] -> 0x2f
[223] -> 0x2f
[224] -> 0x00
```
Without TCXO:
```bash
[  0] -> 0x51
[222] -> 0xff
[223] -> 0xff
[224] -> 0x3f
```
```bash
$ hackrf_info
hackrf_info version: 2021.03.1
libhackrf version: 2021.03.1 (0.6)
Found HackRF
Index: 0
Serial number: 0000000000000000223c69dc268c5d4f
Board ID Number: 2 (HackRF One)
Firmware Version: 2021.03.1 (API:1.04)
Part ID Number: 0xa000cb3c 0x0065475a
```
```

miek, 2021-05-03:
`0x41` in register 0 means that the SI5351C is detecting a clock signal on CLKIN (from the TCXO) but is unable to lock the PLL from that signal. This is because it's only set up for a 10MHz input.

wizarDD03, 2021-05-04:
I see from [here](https://github.com/mossmann/hackrf/wiki/Clocking), the Hackrf can support up to 40MHz on CLKIN with division. The [wiki](https://github.com/mossmann/hackrf/wiki/HackRF-One) says to only use 10 MHz unless you modify the firmware to support this. I'm assuming the modification to firmware is setting the correct division for the selected CLKIN frequency.

Having briefly read the [SI5351C register mapping datasheet](https://www.silabs.com/documents/public/application-notes/AN619.pdf) I see that setting a division is only needed if using a CLKIN above 40MHz. In theory, can I just set the PLLB_SRC in register 15 to CLKIN? Or is it more complicated than that? Or is it easier just to source a 10MHz TCXO.

Thanks for your help.

gozu42, 2021-05-04:
> Having briefly read the [SI5351C register mapping datasheet](https://www.silabs.com/documents/public/application-notes/AN619.pdf) I see that setting a division is only needed if using a CLKIN above 40MHz. In theory, can I just set the PLLB_SRC in register 15 to CLKIN (assuming im using a 22MHz TCXO)? 
> Or is it more complicated than that? 

aiui you will have to adjust the configuration here: https://github.com/mossmann/hackrf/blob/10b7d67ea5f5f9431a206fb4ff2c405f7adaf544/firmware/common/si5351c.c#L123

but both 22MHz and 22.625MHz seem like a really bad freq to work from there, so its somewhere between "have to adjust even more" or "not reasonably possible".

> Or is it easier just to source a 10MHz TCXO.

probably.

wizarDD03, 2021-05-06:
Cheers for the responses. Agreed ~22MHz is not ideal. Law of least efforts... probably just buy a 10MHz. Thanks for your help

closed 2021-05-06 by wizarDD03