greatscottgadgets/hackrf #886: HackRF One not detected via USB port

ChrisTopia, 2021-05-08:
### Steps to reproduce
1. Plug in HackRF One into PC via USB port
2. Enter DFU mode and observe 3V3 and RF lights illuminated
3. Verify device is connected using Windows 10 Device Manager or lsusb/dmesg in Linux (non-virtual)

### Expected behaviour
To see HackRF One listed as a USB device

### Actual behaviour
HackRF One is never detected.

```
I've tried multiple USB ports, multiple USB cables, multiple computers, and multiple operating systems.  Nothing ever shows up.  I've never been able to get this thing running and have never used it with a portapack or anything like that.  Today I decided to open the device up to check for damaged components near the USB connector, but everything looks fine.  It's been very frustrating since I'm a college student and had to save up for months to purchase the device.  I'm hoping to get some advice on what steps to take next.
```

straithe, 2021-05-09:
For each of the operating systems you tried, were they installed directly on the machine or were you running them in VMs?

ChrisTopia, 2021-05-09:
> 
> 
> For each of the operating systems you tried, were they installed directly on the machine or were you running them in VMs?

Both Windows 10 and Linux were installed directly.

gozu42, 2021-05-09:
what happens if you just ... 
* use a linux
* run "sudo dmesg -Tw" in one window
* and plug in the hackRF (without any button-pressing)

usualy DFU mode should not be required, thats a backup/debricking ritual in case your device doesnt have any (working) firmware (yet).

straithe, 2021-05-09:
Can you please post a picture of your HackRF?

straithe, 2021-06-14:
I'm going to close this as there hasn't been a response in a while, but please re-open this issue or open a new one if you still need assistance.

closed 2021-06-14 by straithe

npepper1, 2022-08-17:
I'm having the same issue... what do you need a pic of?

straithe, 2022-08-17:
Please open a new issue.