greatscottgadgets/hackrf #887: Updated firmware without updating CPLD. Now bricked?

justinabrahms, 2021-05-11:
### Steps to reproduce
1. Get a new hackrf one
2. Install hackrf via apt (ubuntu)
3. hackrf_info returns results. I think it was 2018, iirc. It found the board.
4. Update with the latest firmware
5. didn't see the instructions to update CPLD until after. 
6. Can no longer find the hackrf board. USB light doesn't turn on when plugged in (did previously)
7. Tried compiling tools from source hoping that'd work, but no luck.
8. flashed it with dfu, but that doesn't make the usb light turn on.

Lights currently on:

3v3, 1v8, rf. All others are out.

### Expected behaviour
Tell us what you expect should happen

### Actual behaviour
Tell us what happens instead

### Version information
**Operating system**: Ubuntu 20.04

**hackrf_info output:**

hackrf_info version: 2021.03.1
libhackrf version: 2021.03.1 (0.6)
No HackRF boards found.

If you are reporting a problem that involves third party software 
(GNU Radio, Gqrx, etc), please report the version here.

### Output
```
$ dfu-util --device 1fc9:000c --alt 0 --download hackrf_one_usb.dfu
dfu-util 0.9

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2016 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

dfu-util: DFU suffix CRC does not match
dfu-util: A valid DFU suffix will be required in a future dfu-util release!!!
Opening DFU capable USB device...
ID 1fc9:000c
Run-time device DFU version 0100
Claiming USB DFU Interface...
Setting Alternate Setting #0 ...
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 0100
Device returned transfer size 2048
Copying data from PC to DFU device
Download	[=========================] 100%        35508 bytes
Download done.
dfu-util: unable to read DFU status after completion

```

straithe, 2021-05-11:
Did you follow the DFU instructions in https://github.com/mossmann/hackrf/wiki/Updating-Firmware#only-if-necessary-dfu-boot ? One of the steps is to hold down the DFU button while powering the HackRF on or while pressing and releasing the RESET button.

If that works I suggest downloading our[ newest release](https://github.com/mossmann/hackrf/releases/tag/v2021.03.1) directly instead of using the Ubuntu apt source. The latest release manages CPLD so you don't have to update it separately.

justinabrahms, 2021-05-11:
[image]

No change.

justinabrahms, 2021-05-11:
[image]
wrong screenshot. sorry.

straithe, 2021-05-11:
The USB LED lights up after the host configures the USB device, so there could be something going on on the host side that is preventing USB configuration. Do you have Ubuntu running in a VM, or is it running directly on the machine?

justinabrahms, 2021-05-12:
It's running directly on the screen. Before I flashed the updated firmware, the USB light was on.

justinabrahms, 2021-05-12:
It's running directly on the machine. Before I flashed the updated firmware, the USB light was on. Also confirmed that a restart of the host machine doesn't fix the issue.

justinabrahms, 2021-05-12:
Just confirmed that putting it on another machine makes the USB light come on. I'll check my udev rules..

straithe, 2021-05-12:
Let me know how that works out!

justinabrahms, 2021-05-13:
I can't seem to reproduce the success, which is frustrating. I'll continue to try on my windows box. I can say that when I plug in the device on my linux machine, I get no output from `udevadm monitor`, though I do get output when I plug and unplug my mouse.  Still trying to figure out how I can better debug this.

justinabrahms, 2021-05-16:
This is now solved. The issue, to the best I can determine, was that I was using an outdated hackrf_info when contacting a hackrf one which had more recent firmware. Updating to the latest hackrf host software made things better.  Thanks!

closed 2021-05-16 by justinabrahms