greatscottgadgets/hackrf #888: hackrf one power issue

SadSamUK, 2021-05-17:
### Steps to reproduce
1. Plug into any USB port device seen/then not seen
2. After 5 seconds all leds flashing
3. Lower corner next to USB input gets very hot

### Expected behaviour
Device did work well until today - that is over the last 5 years or so!

### Actual behaviour
Plug into any USB port on any computer. The device manager shows the device then lost device this repeats continuously.
The lower casing and main PCB get very hot! The casing becomes too hot to hold after 30 secs.

### Version information
Windows 10 

**hackrf_info output:**
Unable to get hackrf to respond.

If you are reporting a problem that involves third party software 
(GNU Radio, Gqrx, etc), please report the version here.

### Output
```
Insert any commandline or build output here
```

straithe, 2021-05-21:
I strongly advise you to stop plugging your HackRF in over the USB port. It is definitely a hardware failure related to the power supply. It will need to be repaired or replaced. If you have a multimeter and you would like to try repairing this on your own, the first step is to verify that there is no continuity between any of the power supplies and ground. Please let me know what you find!

straithe, 2021-06-21:
I'm going to close this as there hasn't been a response in a while, but please re-open this issue or open a new one if you still need assistance.

closed 2021-06-21 by straithe

SadSamUK, 2021-06-23:
I contacted a few people but I'm unable to get this fixed here in the UK.
So, it's gone into the trash bin. Kr, Sammy

straithe, 2021-06-23:
If you haven't chucked it yet, you can contact your reseller and see if they will do a replacement or exchange.

SadSamUK, 2021-06-24:
Many thanks tried that they are not interested. Not to worry I have jumped
up to an Ettus B210. Many thanks.

roadmutt, 2021-06-25:
on mine the light never turns on but the corner still gets hot does that
help at all

straithe, 2021-06-25:
@roadmutt Open a separate issue and I will help you directly.