greatscottgadgets/hackrf #889: Hi phase noise, especially when externally clocked

automeas, 2021-05-18:
### Steps to reproduce
1. Generate any sinusoidal or DC signal with GNUradio (3.7) or with own software
2. Any power
3. almost any frequency even on 2.5G when mixer is bypassed
4. any sample rate from 2 to 10 MHz
5. checked with fw versions v2021.03.1 and v2014.08.1

### Expected behavior
expected phase noise at least as MAX2837 specified

### Actual behavior

external clock
[image]

internal clock
[image]

boomboompsh, 2021-05-19:
My guess is that the phase noise is coming from the DAC (MAX5864) rather than the MAX2837. The phase noise given in the datasheet is probably the phase noise with the MAX2837's TX mixer disabled, since Maxim has absolutely no way of knowing what you are transmitting and thus can't give a spec for when the mixer is enabled.
 Ruling out the DAC entirely requires manually setting registers on the MAX2837 using hackrf-debug. The relevant bit seems to be bit 8 of register 16. Please realize however that I do have almost no idea what I'm doing and have owned a hackrf for a single day. The project I'm working on does need phase noise to be characterized though, so I will probably be making my own measurements soon with my school's lab equipment.

automeas, 2021-05-19:
It is fully reproducible when DC is generated on baseband.
It looks that si5351c produces more noise than Max2837 can ignore. When I play with MAX2837 PLL filter I can reduce phase noise.

boomboompsh, 2021-05-19:
Also, you're measuring dBc, however phase noise is measured as dBc/hz, the total power over a 1 Hz bandwidth. This is a very big difference. This video shows a "phase noise" of -77 dBc turning into -99 dBc/hz after changing the spectrum analyzer settings. https://youtu.be/SOHjFtw0sgo?t=1616

Each point in your graph represents the power over a few kilohertz of spectrum. By assuming that each pixel is exactly one frequency bin, which seems pretty close, I got that the actual phase noise from your data is 33dB lower, or -83dB.

automeas, 2021-05-19:
For noise-like signals, where power is distributed without much spurs and average detector is used the dBc and dbc/hz is almost the same.
Anyway USRP b210 has much better performance although it is low-end device. At least it's spectrum has descent only, no hill when externally clocked.

avian2, 2021-05-19:
>It looks that si5351c produces more noise than Max2837 can ignore. 

I have been recently exploring phase noise in the *received* signal. The following was calculated from the received signal at 1 GHz using the HackRF's internal clock. Using sampling frequencies of 8 MHz (integer divider mode) and 2.56 MHz (fractional divider). It seems very similar to your measurements for transmitted signal.

[image]

The SI5351 datasheet specifies 150 ps peak-to-peak maximum cycle-to-cycle jitter. As far as I can tell this amount of jitter cannot account for the phase noise I'm seeing at 100 kHz offset.

mossmann, 2021-05-21:
This is a great discussion! I think it is absolutely correct to suspect the Si5351C clock generator as the principal source of phase noise within HackRF One, but it certainly isn't the only contributor. The Si5351C is not specifically intended for radio usage, so its datasheet does not specify phase noise. It does indicate maximum jitter which affects phase noise, but my understanding is that it is not necessarily correct to draw very specific conclusions about phase noise from a jitter specification.

I selected Si5351C because it helped me maximize functionality at low cost. My goal was to make HackRF One highly useful for a wide variety of applications, not to guarantee high performance for any particular application.

automeas, 2021-05-23:
Hi, Michael. I'm not very experienced in PLL synthesizers, but use common cases. I did not find MAX2837 PLL filter calculations, just see that HackRF repeats reference schematic in PLL filter. I increased C114 to 100nF to make PLL filter narrow, assuming that this chain is negative feedback of PLL's OpAmp. It reduces phase noise by ~10 dB when tuned to 2,5 G and there is no mixer contribution.  Si5351C does not have external PLL filter. Possibly it has internal settings of PLL filter or PLL chain gains. Maybe someone has the reference picture of TX CW with surroundings to see regular phase  noise at 50-100K offset?
When we say that 150ps is not much do we account that it will be multiplied to fractional divider when projected to target RF 1G or higher?

mossmann, 2021-06-04:
> When we say that 150ps is not much do we account that it will be multiplied to fractional divider when projected to target RF 1GHz or higher if not suppressed?

I knew of the use as a reference to frequency synthesizers when I selected the Si5351C, but the selection was not made based on quantitative measurements.

As your original concern (that HackRF One output phase noise is higher than the MAX2837 documentation indicates) has been answered (additional phase noise comes from the Si5351C), I'm closing this issue. I invite you to join us on Discord if you would like to have further conversations about these things: https://discord.gg/rsfMw3rsU8

closed 2021-06-04 by mossmann