greatscottgadgets/hackrf #892: Quadrature difference with two HackRF One

Michel0001, 2021-05-26:
Hello, I have two hackrf one with 2017.02.1 software, one of which has ghost frequencies opposite the center frequency.

The difference I may have noticed is an abnormal IQ distribution (Red).

[image]

Is it a hardware or software problem? What would be at the origin of this abnormal IQ distribution?
Thanks.

straithe, 2021-05-31:
Do you still observe this problem after updating to [v2021.03.01](https://github.com/mossmann/hackrf/releases/tag/v2021.03.1)?

Michel0001, 2021-06-02:
I've upgrade firmware to v2021.03.01 and it's the same.
I think it may come from a bad reference for the ADC

[image]

straithe, 2021-06-13:
Can you please post the flow graph that you used to generate your initial set of graphs with both the red and the blue plotted?

Michel0001, 2021-06-16:
[image]
Hello, I think there is a difference in the components. Where can I find the test points  ​​and measured values ?

vladisslav2011, 2021-12-26:
> Is it a hardware or software problem?

Looks like hardware problem. Solder joint on SGPIO2 or broken resistor...

Michel0001, 2022-01-19:
Thanks for the answer, I checked between BANK2F3M9 (74) and SGPIO2 (62) and I have the R89 resistor of 39 Ohms.

vladisslav2011, 2022-01-19:
Measure the resistor to check, that it is not broken. Check both BANK2F3M9 (74) and SGPIO2 (62) for connection to VCC. There should be either connection, that should not exist or missing connection, that should exist. 
Your constellation is showing, that there should be connection between BANK2F3M9 (74) and VCC or SGPIO2 (62) and VCC maybe through surrounding pins. Or broken resistor/traces/bad soldering between BANK2F3M9 (74) and SGPIO2 (62) in case, there is pull-up enabled on SGPIO2 .

straithe, 2022-07-14:
@Michel0001 did you try the suggestion from @vladisslav2011 ?

closed 2022-08-15 by straithe