greatscottgadgets/hackrf #893: hackrf one broken

smlyclts, 2021-05-30:
### Steps to reproduce
1. not connecting to computer
2. only 3.3v led is on
3. not connecting to pc in dfu mode
4. 

### Expected behaviour
HackRF should be detected in PC. USB LED should glow

### Actual behaviour
only 3.3v led light is on

### Version information
windows 10, Linux

**hackrf_info output:**
Unable to connect to computer, hackrf info unknown

If you are reporting a problem that involves third party software 
(GNU Radio, Gqrx, etc), please report the version here.

### Output
I am using clone hackrf one portapack h1 i connect power with power bank but portapack screen turned on and off

only 3.3v led light is on, i disconnected the portapack from hackrf and plugged it into the computer

this is how it shows up in device manager "unknown USB device (device descriptor request failed)"

i changed cable it because the  might be damaged, i changed the power supply, same result

I measured with the help of a voltmeter as in the photo

p3GND AND P8VCC = 3.29-3.3 volts

P3GND AND P1 1V8= 0.01volt

What should I do, will this device work?

[image]

[image]

straithe, 2021-05-31:
I suggest replacing or removing the TVS diode (U15) and seeing if that helps!

smlyclts, 2021-06-20:
> TVS diyot (U15)

Where is the U15 diode? Where can I find a detailed dataset that will show the location of other components

smlyclts, 2021-06-22:
when I examined the u15 I saw that it is usb (D - and D+), when I plug in the portapack it only gets DC (- and +) current with powerbank. I don't think it will work if I dismantle u15
[image]

straithe, 2021-06-28:
Removing U15 does not remove the connection to the microcontroller. The connections you've highlighted don't just go up to U15, they also go right to R58 and R57 which lead to the microcontroller through 0 Ohm resistors which just make them wires. Please do try the recommended solution of removing or replacing U15 and let me know the results.

straithe, 2021-08-23:
I'm going to close this as there hasn't been a response in a while, but please re-open this issue or open a new one if you still need assistance.

smlyclts, 2021-09-20:
[image]
 I took out the u15, the 3.3v led light is on, the computer is not responding, the portapack is not working

straithe, 2021-09-27:
Conversation continues in #958 .

closed 2021-09-27 by straithe