greatscottgadgets/hackrf #896: Erroneous Resource busy error when controlling multiple hackrf

dails08, 2021-06-07:
### Steps to reproduce
1. Connect 2 hackrfs to a single computer
2. Transmit with one
3. Sweep with the other

### Expected behaviour
Normal hackrf sweep output from the receiving hackrf

### Actual behaviour
The hackrf_sweep output will begin with "hackrf_open() failed: Resource budy (-1000)" error, but the output will continue with the expected output.  The Tx led on the transmitting hackrf is lit and the Rx led on the receiving hackrf is lit.

This is a concern because when qspectrumanalyzer tried to run the sweep command for the receiving hackrf, it doesn't work and I think it's the initial error line that causes it to halt, though the outputs from the sweep command do seem to be being piped to the qspectrumanalyzer process as they're not sent to stdout.

### Version information
**Operating system**: Ubuntu 20.04.2 LTS 64 bit

**hackrf_info output:**
`
hackrf_info version: git-10b7d67
libhackrf version: git-10b7d67 (0.6)
Found HackRF
Index: 0
Serial number: 000000000000000075b068dc31917307
Board ID Number: 2 (HackRF One)
Firmware Version: git-6e5cbda (API:1.04)
Part ID Number: 0xa000cb3c 0x00694761

Found HackRF
Index: 1
Serial number: 0000000000000000088869dc38173a1b
Board ID Number: 2 (HackRF One)
Firmware Version: 2021.03.1 (API:1.04)
Part ID Number: 0xa000cb3c 0x006f4349
`

If you are reporting a problem that involves third party software 
(GNU Radio, Gqrx, etc), please report the version here.

### Output
```
(transmit command, adjust the serial number suffix for your device)
hackrf_transfer -d$(hackrf_info | grep 7307 | cut -d : -f 2) -c 64 -f 900000000
call hackrf_set_sample_rate(10000000 Hz/10.000 MHz)
call hackrf_set_hw_sync_mode(0)
call hackrf_set_freq(900000000 Hz/900.000 MHz)
Stop with Ctrl-C
19.4 MiB / 1.000 sec = 19.4 MiB/second
19.7 MiB / 1.000 sec = 19.7 MiB/second
18.6 MiB / 1.000 sec = 18.6 MiB/second
...

(receive command, adjust the serial number suffix for your device)
hackrf_sweep -f 0:1000 -d$(hackrf_info | grep 3a1b | cut -d : -f 2)
hackrf_open() failed: Resource busy (-1000)
call hackrf_sample_rate_set(20.000 MHz)
call hackrf_baseband_filter_bandwidth_set(15.000 MHz)
Sweeping from 0 MHz to 1000 MHz
Stop with Ctrl-C
2021-06-07, 10:04:28.296951, 0, 5000000, 1000000.00, 20, -6.86, -19.13, -36.84, -46.69, -56.54
2021-06-07, 10:04:28.296951, 10000000, 15000000, 1000000.00, 20, -29.33, -41.44, -53.17, -53.29, -56.39
2021-06-07, 10:04:28.296951, 5000000, 10000000, 1000000.00, 20, -54.03, -52.20, -54.18, -66.13, -59.13
...

(qspectrumanalyzer console output)
$ qspectrumanalyzer (set the read from non-transmitting hackrf)
hackrf_sweep params:
{'bin_size': 1000.0,
 'crop': 0,
 'device': 0,
 'gain': 40,
 'hops': 0,
 'interval': 0,
 'lna_gain': 16,
 'ppm': 0,
 'sample_rate': 20000000.0,
 'single_shot': False,
 'start_freq': 0.0,
 'stop_freq': 1000.0,
 'vga_gain': 24}

hackrf_open() failed: Resource busy (-1000)
Usage:
	[-h] # this help
	[-d serial_number] # Serial number of desired HackRF
	[-a amp_enable] # RX RF amplifier 1=Enable, 0=Disable
	[-f freq_min:freq_max] # minimum and maximum frequencies in MHz
	[-p antenna_enable] # Antenna port power, 1=Enable, 0=Disable
	[-l gain_db] # RX LNA (IF) gain, 0-40dB, 8dB steps
	[-g gain_db] # RX VGA (baseband) gain, 0-62dB, 2dB steps
	[-n num_samples] # Number of samples per frequency, 8192-4294967296
	[-w bin_width] # FFT bin width (frequency resolution) in Hz
	[-1] # one shot mode
	[-N num_sweeps] # Number of sweeps to perform
	[-B] # binary output
	[-I] # binary inverse FFT output
	-r filename # output file

Output fields:
	date, time, hz_low, hz_high, hz_bin_width, num_samples, dB, dB, . . .

```

dails08, 2021-06-07:
Similarly, if I start the qspectrumanalyzer sweep first and then try to transmit, I get a resource busy error on the transmitting hackrf.  

After experimenting a bit more, it looks like qspectrumanalyzer is not respecting the Device argument and just uses the first hackrf reported in hackrf_info. If I start sweeping with the first hackrf reported in hackrf_info and then transmit with the second, it seems to work.  If someone else can replicate that, I think we can close this, but it's helpful to know that it's probably not a hackrf problem.

miek, 2021-06-18:
> After experimenting a bit more, it looks like qspectrumanalyzer is not respecting the Device argument and just uses the first hackrf reported in hackrf_info.

Instead you can use the 'Additional parameters' field in QSpectrumAnalyzer to pass the `-d` parameter to `hackrf_sweep` to specify the device serial number.

Also, you may be getting extra device busy errors from the `hackrf_info | grep | cut` call in the commands you pasted since it'll try to open all devices, including any already transmitting/receiving. I would remove that and specify the serial number directly.

> If I start sweeping with the first hackrf reported in hackrf_info and then transmit with the second, it seems to work. If someone else can replicate that, I think we can close this, but it's helpful to know that it's probably not a hackrf problem.

Yep, that also works for me.

closed 2021-06-18 by miek