greatscottgadgets/hackrf #897: Single shot acquisition hackrf

QuentinDem98, 2021-06-14:
Hi guys, 

I currently work with the HackRF One in spectrum analyzer mode and I am trying to make a single shot acquisition in order to get the frequency/magnitude values at T time but I don't know how to do this using GNURadio 3.8. If someone has an idea, I would be greatful about his explanations. 

Thanks for your reply

### Expected behaviour
Make a single shot acquisition like the "single" mode on oscilloscopes

### Actual behaviour
Runs infinitely but I can stop the acquisition with the stop button of the QT GUI Frequency Sink

### Version information
**Operating system / GNURadio**: Windows 10 / 3.8

straithe, 2021-06-14:
We do provide a tool `hackrf_sweep` that you can use from the command line to get a single shot acquisition. To do this you would want to type `hackrf_sweep -1' on the command line. 

If you want a tool with a GUI to get a single shot you might want to try qspectrum analyzer: https://github.com/xmikos/qspectrumanalyzer . I suspect this tool is the best fit for what you are wanting. 

Otherwise, I suggest checking out GNU Radio tutorials or engaging that community.

QuentinDem98, 2021-06-15:
Hi Straithe, 

Thanks a lot. `hackrf_sweep -1` is perfect for what i want to do. 
I close the issue as your reply solved it.

closed 2021-06-15 by QuentinDem98