greatscottgadgets/hackrf #899: Opera Cake board id is in 2 different formats in hack_info and hackrf_operacake

yhetti, 2021-06-15:
`hackrf_info` reports the Opera Cake board id in hex, and `hackrf_operacake -o` expect a numerical value. One of these should be changed so that they use the same format.

mossmann, 2021-06-15:
I2C addresses are confusing, and I don't see any reason to expose the user to them. I'm thinking that all command-line tools should number Opera Cakes 0-7 (0 being the default with no jumpers installed).

miek, 2021-07-14:
> I2C addresses are confusing, and I don't see any reason to expose the user to them. I'm thinking that all command-line tools should number Opera Cakes 0-7 (0 being the default with no jumpers installed).

I started looking into this and hit a slight issue: currently the CLI tools treat `0` as an invalid address in `hackrf_get_operacake_boards`, so if someone used the new firmware with old host tools `hackrf_opercake -l` would not list a board with the default address.

I don't think there's a clean way to deal with this, so I'm leaning towards just accepting the breakage and instructing users to install the new release to use Opera Cake. This is probably a good idea anyway so we don't have to mix I2C addresses into documentation.

mossmann, 2021-07-14:
Yeah, I'm okay with that breakage too as Opera Cake is still in pre-release status.

commit 5e2e06b6 (2021-07-15): hackrf_info: display Opera Cake addresses in decimal

Fixes #899
closed 2021-07-15 by mossmann