greatscottgadgets/hackrf #901: I my u21 amp chip has stopped working what is th part name that I can order it on line

markl17, 2021-06-17:
### Steps to reproduce
1.the u21 chip or preamp stoped working the output voltage is not .014 v which is I think 20db
2.
3.

fixed once I find the chip name

### Actual behaviour
not enough voltage the amo chip is dead

### Version information
**Operating system**: n/a

**hackrf_info output:**

If you are reporting a problem that involves third party software 
(GNU Radio, Gqrx, etc), please report the version here.

### Output
```
Insert any commandline or build output here
```

markl17, 2021-06-18:
Hi as I mentioned u21 that you describe has 8 pins mine 6

mossmann, 2021-06-18:
U21 is a voltage regulator, not an amplifier. Based on your previous issue, I believe you are actually concerned about U25, the transmit amplifier. U25 is Avago part number MGA-81563. This is an older part, but it continues to be available at [some distributors](https://octopart.com/search?q=MGA-81563). This part number is documented in the [KiCad schematic](https://github.com/mossmann/hackrf/tree/master/hardware/hackrf-one) and also in `doc/hardware/hackrf-one-bom.csv` in the [latest release package](https://github.com/mossmann/hackrf/releases/latest).

Based on the information you have provided, I am not confident that you actually have a problem with U25, but I believe this is the part number you are looking for.

markl17, 2021-06-18:
@ @mossmann 
Thank you so much I will look into that

markl17, 2021-06-20:
630-MGA-81563-TR1G I think this is what you are talking about it is a 14 db upto 6ghz but not available qt mouser 
also is there a way to make hackrf go to 10 ghz is there a plan to make one that can go that far anything higher than that

markl17, 2021-06-20:
only one place out of all those on the list have them mouser doesnt cary it either how does one solder and unsolder such a small part without burning it or something

markl17, 2021-06-20:
https://www.berkeleynucleonics.com/model-825
this thing cost above 12k
astronamical 
also how does one create a complex wave of let say two at 500 and 700 mhz and 1.3 ghz 2.3 ghz 3.2 ghz 4.4 ghz 5.5 ghz and 6.6 ghz as an example that they all should bee in their rite timing I mean their wave length gets smaller and smaller yet can they all be broadcast a a complex pattern if that makes cense in a burst i.e 1.5 m 42cm 25cm 12cm 9 cm 7.1 cm 5.9cm and 5.2 cm or there about so that they should appear tucked in one from the other so to say it is a pattern question

markl17, 2021-06-20:
[image]
this is what this guy called u21 which I think is really u25

gbuzogany, 2021-06-21:
Red arrow = U25 = MGA-81563
Blue arrow = U13 = MGA-81563

As mossmann said, the electrical schematics and layout files are available in the hackrf repository. Take a look there, it may help you with the troubleshooting.

straithe, 2021-06-21:
@mossmann and @gbuzogany are correct in the part numbers provided. I am closing this issue as the initial question about part numbers has been resolved.

closed 2021-06-21 by straithe