greatscottgadgets/hackrf #910: Docs image filenames too long for typical ext4 folder structure

kuremu, 2021-07-08:
### Steps to reproduce
1. be using an ext4 partition for /home
2. mkdir -p /home/test/devel/ && cd /home/test/devel
3. git clone https://github.com/mossmann/hackrf

### Expected behaviour
A cloned repository with successful checkout

### Actual behaviour
A cloned repository with unsuccessful checkout

### Output
```
> git clone https://github.com/mossmann/hackrf
Cloning into 'hackrl'...
...
error: unable to create file docs/images/68747470733a2f2f7261772e6769746875622e636f6d2f6d6f73736d616e6e2f6861636b72662f6d61737465722f646f632f77696b692f696d616765732f6261736562616e642d66696c7465722f6d6178323833372d316d373562772d61742d326d2e706e67.png: File name too long
error: unable to create file docs/images/68747470733a2f2f7261772e6769746875622e636f6d2f6d6f73736d616e6e2f6861636b72662f6d61737465722f646f632f77696b692f696d616765732f6261736562616e642d66696c7465722f6d6178323833372d316d373562772d61742d386d2e706e67.png: File name too long
fatal: unable to checkout working tree
warning: Clone succeeded, but checkout failed.
```

### Version information
Linux, ext4 partition

gozu42, 2021-07-08:
can not confirm this one. 
works just fine on ext4 here, and considering it is "only" 208 chars long, there even is some room to grow.

can you please post output for "tune2fs -l /dev/whatever" for that filesystem?

straithe, 2021-07-08:
Thank you for the notice, @kuremu I'll put in a PR to fix those names ASAP.

PR #911 (2021-07-15): updatepicturenames
Closes #910
closed 2021-07-15 by mossmann