greatscottgadgets/hackrf #919: Will you ever post compiled copies of the tools?

Videogamer555, 2021-07-18:
You post compiled copies of the firmware, and in the same ZIP file you post source code (but not EXEs and DLLs) of the tools. Why? If you are already taking the time to compile the firmware, why not just compile the tools too?

dmaltsiniotis, 2021-07-19:
Hi @Videogamer555,

If you're just looking for the latest pre-compiled binaries of the HackRF tools, the PothosSDR project bundles those together with all the required drivers. If [have a look at the HackRF docs here ](https://hackrf.readthedocs.io/en/latest/operating_system_tips.html#windows) they state that:
> Binaries are provided as part of the PothosSDR project, they can be downloaded [here](https://downloads.myriadrf.org/builds/PothosSDR/).

Hope this helps,

-DM

straithe, 2021-07-21:
It is not something we have resources to support at this time, especially when third-party groups like PothosSDR do a great job of creating these for us (as @dmaltsiniotis states).

closed 2021-07-21 by straithe