greatscottgadgets/hackrf #920: The version of compiled HackRF Tools that I'm using from GnuRadio doesn't work.

Videogamer555, 2021-07-18:
### Steps to reproduce
Download the latest Windows version of GNU Radio from http://www.gcndevelopment.com/gnuradio/downloads/installers/v3.9.0.0/gnuradio_3.9.0.0beta_win64.zip
Use a command line to run the HackRF in receive mode (I used the following command line)
hackrf_transfer.exe -w -f 97300000 -s 1.75 -p 0 -a 0 -l 0 -g 0 -n 17500000

### Expected behaviour
It should tune to 97.3MHz, digitize the signal with a 1.75MSPS sample rate, use no amplification, and record for exactly 10 seconds.

### Actual behaviour
It doesn't start recording. Instead it gives the following error.
hackrf_set_sample_rate() failed: Input/Output Error (-1000)

### Version information
**Operating system**: Latest version of Windows 10

**hackrf_info output:**
hackrf_info version: git-cc7f599
libhackrf version: git-cc7f599 (0.5)
Found HackRF
Index: 0
Serial number: 000000000000000057b068dc224b9e63
Board ID Number: 2 (HackRF One)
Firmware Version: 2018.01.1 (API:1.02)
Part ID Number: 0xa000cb3c 0x00484762
hackrf_close() failed: Input/Output Error (-1000)

I'm using the version of HackRF Tools that comes with the latest Windows version of GNURadio, which is gnuradio_3.9.0.0beta_win64 which I downloaded from the link in the first section ("Steps to Reproduce" section) of this bug report.

dmaltsiniotis, 2021-07-18:
I may be off here, but I don't think the hackrf actually supports a sample rate that low: https://hackrf.readthedocs.io/en/latest/tips_tricks.html?highlight=sample%20rate#sampling-rate-and-baseband-filters

>Using a sampling rate of less than 8MHz is not recommended. Partly, this is because the MAX5864 (ADC/DAC chip) isn’t specified to operate at less than 8MHz, and therefore, no promises are made by Maxim about how it performs. But more importantly, the baseband filter in the MAX2837 has a minimum bandwidth of 1.75MHz. It can’t provide enough filtering at 2MHz sampling rate to remove substantial signal energy in adjacent spectrum (more than +/-1MHz from the tuned frequency).

xaki23, 2021-07-18:
another note from the sideline:
```
hackrf_info version: git-cc7f599
```
this is a 2017 version.

```
Firmware Version: 2018.01.1 (API:1.02)
```
this isnt much newer.

please update to a current version of hackrf tooling and firmware.

Videogamer555, 2021-07-18:
> 
> 
> I may be off here, but I don't think the hackrf actually supports a sample rate that low: https://hackrf.readthedocs.io/en/latest/tips_tricks.html?highlight=sample%20rate#sampling-rate-and-baseband-filters so error may actually be expected behavior.
> 
> > Using a sampling rate of less than 8MHz is not recommended. Partly, this is because the MAX5864 (ADC/DAC chip) isn’t specified to operate at less than 8MHz, and therefore, no promises are made by Maxim about how it performs. But more importantly, the baseband filter in the MAX2837 has a minimum bandwidth of 1.75MHz. It can’t provide enough filtering at 2MHz sampling rate to remove substantial signal energy in adjacent spectrum (more than +/-1MHz from the tuned frequency).
> 
> EDIT: Additionally, the -s parameter expects the sample rate in Hz (old hackrf_transfer tools)
> 
> > [-s sample_rate_hz] # Set sample rate in Hz (8/10/12.5/16/20MHz)
> 
> And the latest version of the hackrf tools explicitly call out the range as 2-20:
> 
> > [-s sample_rate_hz] # Sample rate in Hz (2-20MHz, default 10MHz).
> 
> so I don't think 1.75 would be valid. Maybe try 2000000?

I thought I was supposed to specify sample rate by MHz, such as 1.75 for 1.75MHz. I guess I was supposed to actually specify the sample rate by Hz, not MHz so 1750000 for 1.75MHz. Still that doesn't explain the error it gives with hackrf_close() failed: Input/Output Error (-1000) when running the hackrf_info software. That sounds like a USB issue to me.

straithe, 2021-07-18:
As suggested by @xaki23, please update your firmware. Many hackrf_sweep updates were made in that release.

Videogamer555, 2021-07-18:
> 
> 
> another note from the sideline:
> 
> ```
> hackrf_info version: git-cc7f599
> ```
> 
> this is a 2017 version.
> 
> ```
> Firmware Version: 2018.01.1 (API:1.02)
> ```
> 
> this isnt much newer.
> 
> please update to a current version of hackrf tooling and firmware.

Which firmware file should I use to update the firmware? The BIN file or the DFU file? There's hackrf_one_usb.bin and hackrf_one_usb.dfu and I don't know which to use, or even the procedure to use. Since it seems hackrf_tools has issues communicating with the device, how will I even use the tools to update the firmware. Also, I assume the DFU button is used during some point in the procedure, but I'm not sure exactly how. Can you point me to a complete n00b tutorial, telling me step by step how to update the firmware?

xaki23, 2021-07-18:
> Still that doesn't explain the error it gives with hackrf_close() failed: Input/Output Error (-1000) when running the hackrf_info software. That sounds like a USB issue to me.

note you are getting the hackrf_close() error on hackrf_info. this looks like something between "irrelevant" and "old tooling/firmware" to me. 

with hackrf_transfer, your error is on hackrf_set_sample_rate(), which could be explained by trying to set the very-much-out-of-range sample rate of 1.75Hz. 

> The BIN file or the DFU file?

DFU update is basicly a recovery strategy when your hackrf doesnt contain any (usable) firmware at all.
for all normal updates, the .bin file should be fine.
i am not familiar with how to do this on windows. recommend you look for a 2021 version of hackrf tooling for windows, verify the version by running hackrf_info, and then update to the matching firmware.

dmaltsiniotis, 2021-07-18:
> I thought I was supposed to specify sample rate by MHz, such as 1.75 for 1.75MHz. I guess I was supposed to actually specify the sample rate by Hz, not MHz so 1750000 for 1.75MHz. Still that doesn't [...]

I think the error you're getting is pretty clear:
>hackrf_set_sample_rate() failed: Input/Output Error (-1000)

The hackrf tools you are using are correctly communicating to the device, because your hackrf_info is working. Try a sample rate of 2MHz or above, and also updating the the latest tools and firmware for a better user experience. There is _extensive_ documentation that the hackrf folks have taken the time to put together that outlines how to upgrade firmware and many other topics located here: https://hackrf.readthedocs.io/en/latest/updating_firmware.html

straithe, 2021-07-21:
Great support by @xaki23 and @dmaltsiniotis! These answers are correct and I am closing this issue as the question has been answered.

closed 2021-07-21 by straithe