greatscottgadgets/hackrf #921: HackRF one with Touchstone pro not recognizing rf analyzer

jhorton845, 2021-07-18:
### Steps to reproduce
1. plug in hackrf one to MacBook pro
2. launch touchstone pro
3. touchstone pro doesn't recognize a spectrum analyzer

### Expected behaviour
software should launch. 
### Actual behaviour
gives error.
### Version information
**Operating system**:
windows 10 via parallels

**hackrf_info output:**

If you are reporting a problem that involves third party software 
(GNU Radio, Gqrx, etc), please report the version here.

### Output
```
Insert any commandline or build output here
```

jhorton845, 2021-07-18:
Im sorry I'm very new to this product and purchased a Hackrf One to use alongside touchstone pro software. I had done some reading before purchase and was aware I may have to do an update in order to use it as a full 1-6k analyzer.. however now I'm feeling I may be a little over my head.. is the update still needed this late in the game? I see the release was from March so I wasn't sure if these were shipping fully updated.. please advise.. I'm just a monitor engineer looking to use this for RF mics and in-ears in concert settings.

dmaltsiniotis, 2021-07-19:
Hi @jhorton845,

I'm going to attempt to maybe point you in the right direction here, but beware that this particular area of GitHub relates the HackRF One device and its native drivers and tools only. You will most likely not get the help you need since it seems that your issue is not with the HackRF One, but with the software you using, "Touchstone Pro", of which I personally have zero experience with. It also seems that you're going through a level of virtualization, which complicates matters.

Finally, your last sentence is a bit confusing, but without knowing what the software you using is for, I can't really speak that. If you're goal is to monitor RF raw signal levels of transmitting devices, that _seems_ like you're on the right path. If your goal is to actually _listen_ to RF mics and in-ears, that's out of my league.

All that being said, there's a few things to check that are "low hanging fruit".

**Question 1: Have you validated that HackRF One device itself works in your virtual machine, independently of the Touchstone Software?**

You mentioned you're using Parallels to run Windows 10. When you plug the device into your host computer, did you enable USB device redirection to send the USB data from your host computer to your virtual machine running in Parallels? This is done under the "Devices" menu in Parallels as outlined here: https://kb.parallels.com/122993

If USB device redirection is enabled, and you were able to select the HackRF One, the next step is to validate that it's working in your virtualized Windows 10 environment. This can be accomplished by installing the HackRF One libraries and tools, and using the command line program: hackrf_info.exe to see if your virtual machine can "talk" to the device. In order to install these libraries and tools, the [documentation located here](https://hackrf.readthedocs.io/en/latest/operating_system_tips.html#windows) states that you can install the latest version of "Pothos SDR", which comes bundled with the HackRF tools.

Once you have installed the Pothos SDR software, you _should_ be able to open a command prompt, and type in hackrf_info and get an output that looks similar to this:
[image]

If that works, then we can assume with a reasonable degree of confidence that the HackRF One is properly working from your Windows 10 environment, and that the issue is specifically related to the Touchstone Pro software. I would try some of the steps above and see if we can at least get hackrf_info.exe to report that it found a HackRF One before diving into Touchstone. We _might_ get lucky here, and by virtue of installing Pothos, Touchstone may start detecting the HackRF One, but again, this purely speculation.

Cheers,

-DM

straithe, 2021-07-21:
@dmaltsiniotis suggestions were excellent! The only thing we can add is that it appears that Touchstone Pro does not have HackRF support. Since Touchstone Pro doesn't support HackRF and this is an issue related to their software I am closing this issue. If you do have other questions you would like to ask please open a new GitHub issue.

closed 2021-07-21 by straithe

straithe, 2021-07-21:
@dmaltsiniotis your first paragraph in your response was so well worded. May I use that exact wording to respond to other issues in the future?

dmaltsiniotis, 2021-07-22:
> 
> 
> @dmaltsiniotis your first paragraph in your response was so well worded. May I use that exact wording to respond to other issues in the future?

@straithe Of course, glad to be of service.

jhorton845, 2021-07-22:
Thanks for all your input guys. I got it all working tho thank you. 

Justin Horton
Monitor Engineer
FOH - Yelawolf 
New York
845.505.7154

straithe, 2021-07-22:
Great! If you end up writing a blog post on how you got HackRF working with Touchstone Pro, let me know! I'd love to read it.

jhorton845, 2021-07-22:
Yeah they have another program called rationalwaves which is how I found out the hackrf was on their list of approved devices. What I ended up doing was resetting the hack RF while I was booted up in parallels and got the laptop to recognize it through windows rather than through OS. Silly me 🤷🏼‍♂️ 

Justin Horton
Monitor Engineer
FOH - Yelawolf 
New York
845.505.7154

straithe, 2021-07-22:
That's great information! Glad you got it to work.