greatscottgadgets/hackrf #923: Update firmware 2021.03.1

yangfan852219770, 2021-07-21:
### Steps to reproduce
1. Update firmware version 2018.01.1, using following commands:
  hackrf_spiflash -w hackrf_one_usb.bin
  hackrf_cpldjtag -x hackrf_cpld_default.xsvf
2. Then, update firmware version 2021.03.1, using following command:
  hackrf_spiflash -w hackrf_one_usb.bin

### Expected behaviour
I hope change the hackrf firmware version into 2021.03.1

### Actual behaviour
Yeah, there are acutually some changes, such as  **libhackrf version** and **Firmware Version**, but the **hackrf_info version** is still 2018.01.1. I don't know whether it is a success or a failure.

### Version information
**Operating system**: Ubuntu 20.04 LTS

**hackrf_info output:**
hackrf_info version: 2018.01.1
libhackrf version: 2021.03.1 (0.6)
Found HackRF
Index: 0
Serial number: 0000000000000000644064dc262a36cd
Board ID Number: 2 (HackRF One)
Firmware Version: 2021.03.1 (API:1.04)
Part ID Number: 0xa000cb3c 0x0069475e

straithe, 2021-07-21:
You do have an old version of hackrf command line tools installed. This does not have anything to do with firmware. You will have to uninstall them. How you uninstall the hackrf command line tools depends on how you installed them.

yangfan852219770, 2021-07-22:
@straithe 
Year, very thank you! I have solved this problem by uninstalling the old hackrf command line tools 2018. It can get normal output.

closed 2021-07-22 by yangfan852219770

APToni, 2021-11-27:
Hi, during update hackrf one from 2018 to 2021 after perform make in folder hackrf_usb 

toni@ubuntu:~/hackrf-2021.03.1/firmware/hackrf_usb/build$ make
[  1%] Performing build step for 'libopencm3_hackrf_usb'
make[3]: *** No targets specified and no makefile found. Stop.
make[2]: *** [CMakeFiles/libopencm3_hackrf_usb.dir/build.make:130: libopencm3_hackrf_usb-prefix/src/libopencm3_hackrf_usb-stamp/libopencm3_hackrf_usb-build] Error 2
make[1]: *** [CMakeFiles/Makefile2:146: CMakeFiles/libopencm3_hackrf_usb.dir/all] Error 2
make: *** [Makefile:103: all] Error 2
toni@ubuntu:~/hackrf-2021.03.1/firmware/hackrf_usb/build$ 

i can not create bin file for upgrade, who knows what is problem? or where is download bin file 2021??