greatscottgadgets/hackrf #925: A newly bought HackRF One is not working - Signal level received in below -80dB

ganeshbhokare, 2021-07-28:
### Steps to reproduce
1. HackRF One with ANT500 antenna
2. GNU RADIO version 3.9 with SoapySDR interface to access HackRF one through the USB port on a desktop
3. Experiment-1: Soapy HackRF source connected to QT GUI FFT Sink - A simple model to see the spectrum of captured RF signal
4. Experiment-1: Implemented FM receiver to capture SM station and demodulate using WBFM  as demonstrated in FM radio receiver

### Expected behavior
Experiment-1: Want to see received power of RF signal in the range -10-20dB as shown in video lessons
https://www.youtube.com/watch?v=zNUCiGVJQo0&t=1653s&ab_channel=PE1PID
Experiment-2: FM receiver plays only noise at audio sink @ 48kHz

### Actual behavior
Experiment-1: Want to see received power of RF signal in the range -10-20dB as shown in video lessons
Experiment-2: FM receiver plays only noise at audio sink @ 48kHz

### Version information
**Operating system**: Ubuntu 20.04 with GNU RADIO version 3.9 with SoapySDR interface

**hackrf_info output:**
hackrf_info version: unknown
libhackrf version: unknown (0.5)
Found HackRF
Index: 0
Serial number: 000000000000000075b068dc32253e07
Board ID Number: 2 (HackRF One)
Firmware Version: 2018.01.1 (API:1.02)
Part ID Number: 0xa000cb3c 0x0066435d

If you are reporting a problem that involves third party software 
(GNU Radio, Gqrx, etc), please report the version here.

GNU RADIO version 3.9 with SoapySDR interface

### Output
[image]

Insert any command line or build output here
Nil.

dmaltsiniotis, 2021-07-28:
Hi @ganeshbhokare,

Before we dive in too deep here, this issue appears to be related to GNU Radio, and not with the HackRF One device itself. You will most likely not find the help you are looking for here since this particular GitHub project page is dedicated to the HackRF one device and its related libraries. Now, that being said, we can try to help you along (I'm sure a there are some folks have played around with GNU Radio here) but no promises :)

From the screenshot you provided, it looks like you've defined a variable called center_freq and intended to tune to 97.6 MHz, but you have not actually linked that variable to the Soapy HackRF Source. The HackRF is _actually_ tuned to 2.0 GHz (well outside the supported frequency of the ANT 500 antenna), and the FFT plot screenshot reflects this. If I tune my HackRF with the ANT 500 antenna attached to 2GHz with no amplification, I get a similar result.

**Try double clicking the Soapy HackRF source and replace the 2GHz tuning frequency with the variable named you defined of "center_freq" and see if that changes the tuning of the device to the range you're expecting.**

Some things to keep in mind when debugging issues:
* If the RX light turns yellow, and hackrf_info is working, and GNU Radio is receiving data, then unless you have reason to specifically suspect hardware damage the HackRF is _most likely_ operating as intended.

* Your hackrf_info is showing the device is responding (albeit with an older firmware version [that should be updated](https://hackrf.readthedocs.io/en/latest/updating_firmware.html#updating-the-spi-flash-firmware)) and the FFT plot shows that GNU Radio is receiving data from the device, including the DC spike in the center.

Full disclosure: I am not directly affiliated with the project, I'm just "some dude" on the internet :)

dmaltsiniotis, 2021-07-28:
One more thing I'll add, if you want to simply check that the device is working and receiving signals, you can use any number of "slightly more user-friendly" SDR applications to visualize the RF spectrum in real-time (and even demodulate some signals like FM radio with a few clicks.) Some example applications that come to mind that support Linux are:
* [GQRX](https://gqrx.dk/)
* [CubicSDR](https://cubicsdr.com/)
* [SDR++](https://github.com/AlexandreRouma/SDRPlusPlus)

This would help rule out issues with the HackRF One device itself as a source of the problem.

ganeshbhokare, 2021-07-29:
Hi Demetri,

Thank you for your email and response to my query. I also appreciate your
help and guidance in this matter.

I understood your point of center frequency of 2GHz. Actually, I bought 2
antennas for HackRF one

   1. ANT-500 and
   2. Second antenna there is no number but it operates from 700 MHz to 5.8
   GHz

I did my experimentation with both antennas and the snapshot I have posted
is for a second antenna operating in the range 700 MHz to 5.8 GHz
for which I set the center frequency to 2GHz just for  testing and to
debug the problem.

For ANT-500, I have set center_freq to multiple stations 98.3MHz, 101MHz
which  are FM radio stations in our area. However, I am still getting
noisy output.
Please see the snapshot below which shows signal level way belo -40dB.

Can you put some light on whether the spectrum of RF signals is correct or
not? It will definitely help to move forward.

[image: HackRF-One-SignalPower-ANT500.png]

Meanwhile, I will explore the options provided by you and will try to debug
HackRF.

   - GQRX <https://gqrx.dk/>
   - CubicSDR <https://cubicsdr.com/>
   - SDR++ <https://github.com/AlexandreRouma/SDRPlusPlus>

Thanks & Best Regards,
*Dr. Ganesh*
*Freelance Consultant,*
*Computer Vision, Deep Learning & Signal Processing*

straithe, 2021-07-31:
Your screenshots are not included when you reply to a GitHub issue via email. You will have to visit the issue on GitHub and add your images there.

ganeshbhokare, 2021-08-02:
Thank you for your email and response to my query. I also appreciate your help and guidance in this matter.

I understood your point of the center frequency of 2GHz. Actually, I bought 2 antennas for HackRF one 
ANT-500 and 
The second antenna there is no number but it operates from 700 MHz to 5.8 GHz 
I did my experimentation with both antennas and the snapshot I have posted is for a second antenna operating in the range 700 MHz to 5.8 GHz
for which I set the center frequency to 2GHz just for testing and to debug the problem.

For ANT-500, I have set center_freq to multiple stations 98.3MHz, 101MHz which  are FM radio stations in our area. However, I am still getting noisy output.
Please see the snapshot below which shows the signal level way below -40dB. 

[image]

Can you put some light on whether the spectrum of RF signals is correct or not? It will definitely help to move forward.
Meanwhile, I will explore the options provided by you and will try to debug HackRF.
GQRX
CubicSDR
SDR++

Thanks & Best Regards,

ganeshbhokare, 2021-08-02:
Hi Straithe,

I uploaded my query to a GitHub issue as you suggested. Please have a look.

Thanks & Best Regards,
*Dr. Ganesh*
*Freelance Consultant,*
*Computer Vision, Deep Learning & Signal Processing*

straithe, 2021-08-20:
Were you able to try one of the alternative pieces of software, like GQRX, to see if your HackRF is working properly?

ganeshbhokare, 2021-08-21:
Dear Straithe,

Yes, I received the HackRf one unit and I tested it and I can catch FM
stations. This unit is working fine.

Thank you very much for your support.

Best Regards,
Dr. Ganesh

closed 2021-08-22 by straithe

pablozl85, 2023-03-27:
hello, I have a problem that my hackrf receives signals and does not send in any way reception tested on flipper zero car and other devices simply does not send a signal does anyone have any ideas for it? thanks in advance

straithe, 2023-03-27:
Please open a new issue @pablozl85