greatscottgadgets/hackrf #926: Using a GPIO for PTT

Gizmolo39, 2021-08-01:
Would it be possible to use a GPIO for switching a PA or antenna switch, when transmitting?

regards

Marcel

gozu42, 2021-08-01:
sure, why not.
there are actualy two GPIOs that get set ON in TX mode already:
```
SCU_PINMUX_LED3      /* GPIO2[8] on P6_12 */
SCU_TX_AMP            /* GPIO2[15] on P5_6 */
```
the easier of the two is probably LED3 (aka "the tx led"), which is available on the unpopulated P5 header.

Gizmolo39, 2021-08-01:
Thanks for your reply!
But unfortunaly every PA and antenna switch I know need the signal to be grounded, say low. So one needs a signal to be set to gnd.

gozu42, 2021-08-01:
a glue circuit would be recommended anyways since it seems like a bad plan to directly connect a gpio to an external PA.
some combination of optocoupler/inverter/fet should do the trick.
details depend on your isolation and sinking requirements.

Gizmolo39, 2021-08-01:
That's what I'm doing at the moment. But driving a FET should not be a challenge ;)