greatscottgadgets/hackrf #927: hackrf_sweep stops outputting data after repeated calls

olilogic77, 2021-08-04:
### Steps to reproduce
1.  run hackrf_sweep from linux command line repeatedly
2.
3.

### Expected behaviour
Tell us what you expect should happen
normal hackrf_sweep output
### Actual behaviour
Tell us what happens instead
output is just
13.4 MiB / 1.000 sec = 13.4 MiB/second
13.4 MiB / 1.000 sec = 13.4 MiB/second
13.4 MiB / 1.000 sec = 13.4 MiB/second
13.4 MiB / 1.000 sec = 13.4 MiB/second
13.1 MiB / 1.000 sec = 13.1 MiB/second

### Version information
**Operating system**:
raspbian

**hackrf_info output:**

If you are reporting a problem that involves third party software 
(GNU Radio, Gqrx, etc), please report the version here.

### Output
```13.4 MiB / 1.000 sec = 13.4 MiB/second
13.4 MiB / 1.000 sec = 13.4 MiB/second
13.4 MiB / 1.000 sec = 13.4 MiB/second
13.4 MiB / 1.000 sec = 13.4 MiB/second
13.1 MiB / 1.000 sec = 13.1 MiB/second

Insert any commandline or build output here
```hackrf_sweep

gozu42, 2021-08-04:
could you please add the hackrf_info output?
if the firmware or tooling are pre-2021 version, please update and try again, there were several fixes to sweeping.
(though none where i would expect "no output")

how are you "stopping" the hackrf_sweep processes in this? 
as in, with a commandline option? or ctrl-c?

olilogic77, 2021-08-04:
thanks so much for your quick reply!

Looks like I should be upgrading.  I'll give that a go.  
Originally I was using -1 for a single sweep for different ranges, but then stopping with ctrl-c to see if that made a difference instead.

hackrf_info version: unknown
libhackrf version: unknown (0.5)
Found HackRF
Index: 0
Serial number: 0000000000000000088869dc2a82261b
Board ID Number: 2 (HackRF One)
Firmware Version: 2017.02.1 (API:1.02)
Part ID Number: 0xa000cb3c 0x005d475a

gozu42, 2021-08-04:
oh. oh yeah. that very much looks like an update is worth a try.

if you can reproduce the problem with 2021 versions, please check if "dmesg -Tw" in a different window reports anything when the problem happens.

miek, 2021-08-04:
This looks a lot like #368 which was fixed by #702 & first released in v2021.03.1

olilogic77, 2021-08-04:
Yes, the #368 issue looks very similar to mine.

I've updated the firmware to 2021, but I don't think the hackrf tools updated.  It looks like the latest version via apt-get is the 2018 version.

So, I updated to the latest version of hackrf tools by compiling from source, rebooted and now I get ...
``` 
call hackrf_sample_rate_set(20.000 MHz)
call hackrf_baseband_filter_bandwidth_set(15.000 MHz)
Sweeping from 0 MHz to 6000 MHz
hackrf_sweep: symbol lookup error: hackrf_sweep: undefined symbol: hackrf_start_rx_sweep
```

Any help to get back to a workable system would be great

Disclaimer: I know my way around a command line but haven't done a lot of compiling in the linux space, so its highly likely there was user error involved.

dmaltsiniotis, 2021-08-04:
My guess is that the old version of libhackrf is still installed system-wide and that's what the tools you compiled are inadvertently using, hence the undefined symbol since the sweep functionality is _relatively_ new. If you've gone so far as to actually compile the tools from source and get those to work, it's only a short step further to compile the library as well (actually, I think it's part of the cmake build process already) and replace the old library on your system with the new one.

I'm not entirely sure what the best way to go about this on Linux would be, my guess would be to apt-get remove the old hackrf libs and tools, and then place your compiled version of the library in it's place?

gozu42, 2021-08-04:
building hackrf tooling from source installs to /usr/local/lib by default, which is not in the default search path on raspbian.
the "pragmatic" solution here is to add "/usr/local/lib" as the first line to "/etc/ld.so.conf" and run "sudo ldconfig". 
(after that, anything installed in /usr/local/lib will be "found first", so be used even if the system-installed lib in "/usr/lib/aarch64-linux-gnu" is still around.)

olilogic77, 2021-08-04:
ok, I think I successfully compiled everything after a bit of a mess I made.  

However, on running hackrf_sweep -1 I get expected output the first couple of times, but then :(
```call hackrf_sample_rate_set(20.000 MHz)
call hackrf_baseband_filter_bandwidth_set(15.000 MHz)
Sweeping from 0 MHz to 6000 MHz
Stop with Ctrl-C
0 total sweeps completed, 0.00 sweeps/second
0 total sweeps completed, 0.00 sweeps/second
0 total sweeps completed, 0.00 sweeps/second
0 total sweeps completed, 0.00 sweeps/second
0 total sweeps completed, 0.00 sweeps/second
0 total sweeps completed, 0.00 sweeps/second
0 total sweeps completed, 0.00 sweeps/second
0 total sweeps completed, 0.00 sweeps/second
0 total sweeps completed, 0.00 sweeps/second
0 total sweeps completed, 0.00 sweeps/second
0 total sweeps completed, 0.00 sweeps/second
0 total sweeps completed, 0.00 sweeps/second
0 total sweeps completed, 0.00 sweeps/second
0 total sweeps completed, 0.00 sweeps/second
0 total sweeps completed, 0.00 sweeps/second
^CCaught signal 2

Exiting...
Total sweeps: 0 in 15.18886 seconds (0.00 sweeps/second)
hackrf_stop_rx() done
hackrf_close() done
hackrf_exit() done
exit
```

```hackrf_info
hackrf_info version: 2021.03.1
libhackrf version: 2021.03.1 (0.6)
Found HackRF
Index: 0
Serial number: 0000000000000000088869dc2a82261b
Board ID Number: 2 (HackRF One)
Firmware Version: 2021.03.1 (API:1.04)
Part ID Number: 0xa000cb3c 0x005d475a
```

straithe, 2021-09-01:
Have you tried running this not on a raspberry pi?