greatscottgadgets/hackrf #929: HackRF failed to accept new firmware, device not found

RoninActual, 2021-08-09:
I just tried to update my portapack to the newest mayhem fork.  After boot it would not enter HackRF mode properly as it showed blue text on the screen and didn't function correctly on my host system.  After following the instructions for a firmware update to vanilla HackRF via spiflash hackrf_info could not recognize the device.  I then executed the DFU procedure, followed by the SpiFlash procedure as prescribed. The HackRF is still not recognized, I'm getting green on the first led between the rest/dfu button and a red led on the 3rd. 

@dominicgs did you ever post that Flash Chip test you mentioned in a previous issue similar to this one?  I'm hoping I don't need a hardware replacement.

### Steps to reproduce
1. SpiFlash to current mayhem firmware
2. SpiFlash to current HackRF firmware
3. DFU boot to current HackRF firmware
4. SpiFlash to current HackRF firmware

### Expected behavior
Fresh firmware

### Actual behavior
hackrf_info doesn't recognize the device

### Version information
**Operating system**: 
DragonOS (Lubuntu)

**hackrf_info output:** 
hackrf_info version: 2021.03.1
libhackrf version: 2021.03.1 (0.6)
No HackRF boards found.

### Output
```
$ dfu-util --device 1fc9:000c --alt 0 --download hackrf_one_usb.dfu
dfu-util 0.9

dfu-util: DFU suffix CRC does not match
dfu-util: A valid DFU suffix will be required in a future dfu-util release!!!
Opening DFU capable USB device...
ID 1fc9:000c
Run-time device DFU version 0100
Claiming USB DFU Interface...
Setting Alternate Setting #0 ...
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 0100
Device returned transfer size 2048
Copying data from PC to DFU device
Download        [=========================] 100%        35508 bytes
Download done.
dfu-util: unable to read DFU status after completion

$ hackrf_spiflash -w hackrf_one_usb.bin
File size 35444 bytes.
hackrf_open() failed: HackRF not found (-5)
```

straithe, 2021-08-09:
If you take the HackRF out of the PortaPack and run the hackrf_info command what happens?

miek, 2021-08-09:
Could you run the dfu command again and then post the output of `hackrf_spiflash --status -v`?

RoninActual, 2021-08-09:
> Could you run the dfu command again and then post the output of `hackrf_spiflash --status -v`?

hackrf_open() failed: HackRF not found (-5)

RoninActual, 2021-08-09:
@straithe you were right, it just needed to be pulled from the portapack.  I ran the DFU boot and spiflash worked.  Thank you guys for responding in literal seconds.  this issue is resolved.

closed 2021-08-09 by RoninActual