greatscottgadgets/hackrf #930: Opera Cake CLI/API cleanup

miek, 2021-08-10:
With our first supported Opera Cake release coming up, I think it makes sense to do any UI/UX/breaking changes sooner rather than later (like #917 for example). As I've been working on #527 & #528 I've found a few other things I think are worth fixing:

- [ ] Deprecate & replace `hackrf_set_operacake_ranges`. I think `hackrf_set_operacake_freq_ranges` would be clearer and also it would be better to pass in a struct* argument rather than an undocumented set of bytes (like https://github.com/miek/hackrf/commit/819f3be5382480b1e813ed03067ff4b82f828d13#diff-e4d18e38ea1b9068f7e9fac04eb7bb2df084d1425fec3992025b893ceab7c529).
- [ ] Let `hackrf_operacake` use a default address of `0` without specifying it manually.
- [ ] Swap the argument order in `hackrf_operacake -f` to <port:min:max>. This would allow the time-switching mode to use `-t <port:dwell>` and make the dwell part optional (defaulting to a global value).
- [ ] Make the main actions of `hackrf_operacake` mutually exclusive. Right now you can mix & match list/gpio-test/settings leading to some confusing behaviour/output.

Do those all sound reasonable?

mossmann, 2021-08-11:
agreed!

mossmann, 2021-08-11:
While I think a struct argument would be great for the host API, I think it is important to continue handling it byte by byte when passing it to or from USB to avoid endianness issues.