greatscottgadgets/hackrf #932: -36db

markl17, 2021-08-12:
### Steps to reproduce
1.a simple radio setup to generate frequency 
2.
3.

### Expected behaviour
Tell us what you expect should happen
The dB should be with out any amplifier should be +10fb

### Actual behaviour
Tell us what happens instead
Instead I get -34db

### Version information
**Operating system**:

**hackrf_info output:**

If you are reporting a problem that involves third party software 
(GNU Radio, Gqrx, etc), please report the version here.

### Output
```
Insert any commandline or build output here
```

mossmann, 2021-08-20:
There are a couple of things misconfigured in the GNU Radio flowgraph:

1. The minimum sample rate supported by HackRF One is 2 Msps (see https://hackrf.readthedocs.io/en/latest/hackrf_one.html#features) but your `samp_rate` is set to 48 ksps. I recommend setting it to 4 Msps (`4e6`) for best results.

2. The maximum frequency that can be generated by the Signal Source without aliasing is half the sample rate, but your Signal Source is configured to generate a frequency of 200 MHz. I suggest that you replace the Signal Source with a Constant Source and set its constant to 1. (The HackRF will convert that constant signal to a 200 MHz sinusoid in the analog domain.)

Additionally you might not be aware of the difference between dB and dBm. If you want an output power of approximately 10 dBm, you should increase the IF gain in the Osmocom Sink to 47 dB which is the maximum. This will maximize the output power but does not guarantee any particular output power level. HackRF One output power is uncalibrated.

These are all topics covered in my video series which I think you may find helpful: https://greatscottgadgets.com/sdr/

markl17, 2021-08-20:
as you see I did all of your changes and got -39db 

[image]
[image]

markl17, 2021-08-22:
Why is this closed I didn’t close it I still have problems

straithe, 2021-08-22:
This particular area of GitHub relates mainly to the HackRF One device and its native drivers and tools only. You will most likely not get the amount help you need since it seems that your issue is not with the HackRF One, but with GNU Radio. As suggested by @mossmann you may want to check out https://greatscottgadgets.com/sdr/ or ask more questions in the GNU Radio community.

markl17, 2021-08-22:
hi I think that one of my amplifier chips is dead how do I find if it's not

markl17, 2021-08-25:
i really think one of my amp chips is blawn

closed 2021-09-10 by straithe