greatscottgadgets/hackrf #933: Trouble getting pentoo to load desktop

KevinLBell, 2021-08-15:
### Steps to reproduce
1. Purchased HackRF One with ANT500 & SMA Antenna Adapter Bundle (on Amazon) 
2. on Intel i7 machine, downloaded Pentoo Full amd64 hardened / version: 2021.0_p20210814
3. burned iso using unetbootin to usb drive , restart computer with usb drive inserted , use load with (verify) 1st Option
4. syslinux.cfg: FAILED during initial screen , all others say: OK
5. after 5 minutes list of :OK, OK , etc... at end says Some checksums failed, press any key to poweroff ...
6. tried load without (verify) 2nd option
7. multi can not connect to server errors, get to Enter new password: (I put a password in), login with pentoo pass: (mine), at prompt type: startx , receive error giving up, unable to connect to X server: Connection refused, server error, then back to prompt

### Expected behaviour
get into desktop environment 

### Actual behaviour
multi errors

### Version information
**Operating system**: Genkernel 4.2.3 (2021-08-14 ..) Linux kernel 5.12.13-pentoo

**hackrf_info output:**
Board ID Number: 2  (HackRF One)
Firmware: 2018.01.1 (API:1.02)
Part ID Number: 0xa000cb2c 0x0054436f

If you are reporting a problem that involves third party software 
(GNU Radio, Gqrx, etc), please report the version here.

### Output
```
Insert any commandline or build output here
```

KevinLBell, 2021-08-15:
note: at 7. I did get an IP address from the router for this device before typing : startx

hookaz, 2021-08-15:
This is not a hackrf issue, it is a pentoo issue. To that I suggest asking on their discord.

closed 2021-08-16 by KevinLBell