greatscottgadgets/hackrf #936: RX or sweep mode wont work on compiled frimware from master branch

best4username, 2021-08-19:
### Steps to reproduce
1. Pull repo
2. submodule init & update
3. cmake & make the frimware
4. hackrf_spiflash -w hackrf_usb/hackrf_usb.bin
5. hackrf_transfer -f 700e6 -l  32 -g 20 -s 10e6  -r -  (stops after 1 sec)
6. hackrf_sweep -f 88:108 ( also stops after 1 sec)

### Expected behaviour
When I flash the compiled one from releases  it work fine, So My compiled one should be the same 

### Actual behaviour
No Output samples When work in rx mode 
**hackrf_sweep**
```
call hackrf_sample_rate_set(20.000 MHz)
call hackrf_baseband_filter_bandwidth_set(15.000 MHz)
Sweeping from 88 MHz to 108 MHz
Stop with Ctrl-C
0 total sweeps completed, 0.00 sweeps/second

Couldn't transfer any data for one second.
```
**hackrf_transfer**
```
call hackrf_sample_rate_set(20.000 MHz)
call hackrf_baseband_filter_bandwidth_set(15.000 MHz)
Sweeping from 88 MHz to 108 MHz
Stop with Ctrl-C
0 total sweeps completed, 0.00 sweeps/second

Couldn't transfer any data for one second.

```

### Version information
**Operating system**:
```
NAME="Manjaro Linux"
ID=manjaro
ID_LIKE=arch
BUILD_ID=rolling
PRETTY_NAME="Manjaro Linux"

```

tried also on **Ubuntu 21**

```
NAME="Ubuntu"
VERSION="21.04 (Hirsute Hippo)"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Ubuntu 21.04"
VERSION_ID="21.04"
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
VERSION_CODENAME=hirsute
```

**hackrf_info output:**

```
hackrf_info version: 2021.03.1
libhackrf version: 2021.03.1 (0.6)
Found HackRF
Index: 0
Serial number: 000000000000000046d067dc324b3747
Board ID Number: 2 (HackRF One)
Firmware Version: git-7047252 (API:1.04)
Part ID Number: 0xa000cb3c 0x0058435f

```
**arm-none-eabi-gcc**
```
Using built-in specs.
COLLECT_GCC=arm-none-eabi-gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/arm-none-eabi/11.2.0/lto-wrapper
Target: arm-none-eabi
Configured with: /build/arm-none-eabi-gcc/src/gcc-11.2.0/configure --target=arm-none-eabi --prefix=/usr --with-sysroot=/usr/arm-none-eabi --with-native-system-header-dir=/include --libexecdir=/usr/lib --enable-languages=c,c++ --enable-plugins --disable-decimal-float --disable-libffi --disable-libgomp --disable-libmudflap --disable-libquadmath --disable-libssp --disable-libstdcxx-pch --disable-nls --disable-shared --disable-threads --disable-tls --with-gnu-as --with-gnu-ld --with-system-zlib --with-newlib --with-headers=/usr/arm-none-eabi/include --with-python-dir=share/gcc-arm-none-eabi --with-gmp --with-mpfr --with-mpc --with-isl --with-libelf --enable-gnu-indirect-function --with-host-libstdcxx='-static-libgcc -Wl,-Bstatic,-lstdc++,-Bdynamic -lm' --with-pkgversion='Arch Repository' --with-bugurl=https://bugs.archlinux.org/ --with-multilib-list=rmprofile
Thread model: single
Supported LTO compression algorithms: zlib zstd
gcc version 11.2.0 (Arch Repository) 
```

best4username, 2021-08-19:
I removed arm-none-eabi-gcc package and donwload the tar from here:
https://developer.arm.com/tools-and-software/open-source-software/developer-tools/gnu-toolchain/gnu-rm/downloads
then install it and try to compile and flash with no sucsses !

straithe, 2021-09-19:
Thank you for the detailed summary and description of the issue you are experiencing! Through some experimentation I've found that the compiler may be one of the factors at play and I suggest trying to compile the firmware again but with the compiler described here: https://github.com/mossmann/hackrf/blob/v2021.03.1/firmware/README

I am still going to do some more investigation into the HackRF code, but hopefully this will get your instance working well while I continue this investigation.

miek, 2021-09-20:
If you had used the same clone to build older versions before building the latest release, this might be caused by #693. Try checking out master, running `make clean` before building and see if that makes any difference.

best4username, 2021-09-23:
Tried all suggests, Without any Successes ??

straithe, 2021-09-23:
Did you uninstall all previous versions of HackRF software that you had on your system before using the suggested compiler to rebuild the firmware?

mossmann, 2021-09-29:
I was able to reproduce this problem with firmware compiled using Ubuntu 21.04's `arm-none-eabi-gcc` (9.2.1) and with the latest release from https://developer.arm.com/tools-and-software/open-source-software/developer-tools/gnu-toolchain/gnu-rm/downloads (release 10.3-2021.07 which is version 10.3.1).

I was not able to reproduce the problem with previous releases from developer.arm.com, so I recommend using release 10-2020-q4-major (10.2.1) until we can determine the cause of the problem.

miek, 2021-10-06:
I was able to reproduce this too, and doing a git-bisect found the first failing commit as e76eace09d7ab77224f62cee84e73b7179730bd3 where we switched to using the M0 core to service the SGPIO peripheral.

I tried changing the M0 core to just blink an LED and that didn't work either, suggesting the M0 wasn't running.

It turned out to be that the M0 firmware image was no longer being merged into the main `.text` section so it didn't get copied to RAM on boot, and `m0_rom_to_ram` would just copy uninitialised RAM to the M0's region.

This should be fixed in #967 which changes `m0_rom_to_ram` to copy the image directly from ROM instead.

commit 7f21c93c (2021-10-13): Copy M0 image directly from ROM, instead of shadow region.

This changes m0_rom_to_ram to read directly from the ROM region, rather
than copying from where .text has been copied to RAM.

Previously the second .text section would be merged with the first one,
so the M0 image would be copied to RAM during `pre_main`. However, on
newer linkers the sections are not merged together so the second .text
section did not get copied to RAM.

fixes #936