greatscottgadgets/hackrf #938: cannot open hackrf device is the device locked somewhere

gaspar1987, 2021-08-23:
How we can repair the to hackrf one repair the pitole usb not work on pc the hackrf has led green 3V3.

Please give me the tool that could fix it if you can help me fix it that would be nice

straithe, 2021-08-23:
I would like to help but I am not certain that I understand what you are asking for. Are you asking for a tool to open the case that goes around a HackRF? Can you provide a picture?

gaspar1987, 2021-08-23:
the problem comes from the buttons when I try to restart it does not restart and this is the image it tells me that I give you .it doesn't plug into my computer### 
[image]

straithe, 2021-08-23:
I don't think I'm familiar with this case. 

What doesn't plug into your computer?

gaspar1987, 2021-08-23:
hello I would like to know if there is a software to reformat it

gaspar1987, 2021-08-23:
thank you for your help

straithe, 2021-08-23:
The documentation for HackRF is here: https://hackrf.readthedocs.io/en/latest/

gaspar1987, 2021-08-23:
what I need to download on I don't understand

straithe, 2021-08-23:
I'm sorry. I still don't understand what you are asking for. All of the HackRF software is available in this repository. All the information you need should be in the documentation that I have linked.

straithe, 2021-09-27:
I'm going to close this as there hasn't been a response in a while, but please re-open this issue or open a new one if you still need assistance.

closed 2021-09-27 by straithe

Janxyxy, 2025-03-18:
[image]
I got this error in AIRSPY SDR when connecting HackRF one.

martinling, 2025-03-18:
@Janxyxy  This normally means that some other program still has the HackRF device open.

If you're still having trouble please open your own issue rather than commenting on old closed ones.