greatscottgadgets/hackrf #940: Usb cable too hot

markl17, 2021-08-29:
### Steps to reproduce
1.plug in usb cable
2.too hot to touch
3.

### Expected behaviour
Tell us what you expect should happen
Normal hackrf turn on
### Actual behaviour
Tell us what happens instead

### Version information
**Operating system**:

**hackrf_info output:**

If you are reporting a problem that involves third party software 
(GNU Radio, Gqrx, etc), please report the version here.

### Output
```al the right led are turned on before screen turned dark it was very dark but Mac claimed it didn’t recognize it 
Insert any commandline or build output here
```
https://youtu.be/Z9OQBgECMnU

porkiedev, 2021-08-29:
Your YouTube video is marked as private so I can't see it.

Does the HackRF work still? Is your computer detecting the HackRF when you plug it in? Can you tx/rx?
USB Cables are literally just cables, they don't get hot unless you have a unprotected short or you're drawing too much current.

Also, did you buy a 'ebay' HackRF, or did you buy a genuine? While ebay versions of the HackRF should be the exact same, their QC is commonly bad, so if it's a new HackRF there may be an issue with the board. Otherwise I would try a different USB cable if you haven't already, and I would try a different USB port on your computer.

Remember to try 1 thing at a time so you know what the actual problem was.
Hope this helps.

straithe, 2021-09-01:
@porkiedev is correct, @markl17 your video is private so we cannot see. As suggested, you should try a new USB cable.

markl17, 2021-09-02:
All usb cords get hot something like resistor got burned when one of the
amps wasn’t plugged in to +5.2

porkiedev, 2021-09-02:
I wasn't aware USB cables had resistors. I know USB- C does.

Nonetheless, a fried resistor would well, fry, and no longer work. The heat should minimally transfer through the cable itself.

But regardless, forgive me but your response is a bit lackluster of information. Again, try a different USB cable and different USB port in your computer. And regarding USB voltage, if you accidentally put more than 5.2v into the hackrf you may have fried some serial connectios resulting in a non functional HackRF, but that still doesn't explain why the USB cable would get so hot. I'm still convinced there is a short somewhere. There could be a short on the hackrf causing too much current draw. Is the hackrf getting warm anywhere when you plug it in but don't use it?

markl17, 2021-09-02:
I tried 4 it’s the board not the cables when too much current goes through
a cheap half amp cable they heat up

markl17, 2021-09-02:
When there’s no resistance on the USB port it’s starts to draw more then it
was designed to

markl17, 2021-09-02:
It’s a chain event amp needs more current it couses the inputting usb to
draw more than it should and burns up

porkiedev, 2021-09-02:
You're right. I know USB cables have resistors I just forgot. Would've helped if I was thinking about current limiting resistors.

Anyways, I have used different usb cables for my hackrf and had no issues. 4 usb cables all containing 250ma current resisting limits is pretty crazy, that's a very low value. Something seems off here but I need to head to bed so I won't be able to help any further for the next few hours. Sorry.

Kind regards, porkiedev

straithe, 2021-09-02:
Can you please post a picture of your board so I can look for shorts, @markl17 ?

markl17, 2021-09-02:
Here are some pictures

markl17, 2021-09-02:
i am trying again
[image]
[image]
[image]
[image]
[image]

straithe, 2021-09-03:
Can you please post one picture of the entire board?

markl17, 2021-09-03:
Sure, here it is.

porkiedev, 2021-09-03:
> Sure, here it is.
You never sent an image..

markl17, 2021-09-03:
[image]

straithe, 2021-09-10:
It sounds like you do have a hardware failure. All Great Scott Gadgets resellers replace defective HackRFs. Unfortunately you don't have HackRF made by Great Scott Gadgets so all I can suggest is contacting the person you purchased from to see if they might give you a replacement.  Good luck!

closed 2021-09-10 by straithe