greatscottgadgets/hackrf #942: HackRF won't respond with either reset or dfu

kwilkins9, 2021-08-31:
So I've been playing with my hackrf and portapack h1 for a few days and it has been great. I started to build the development tools to see what else I can do and the unit shut off.  I unplugged it and it powered on but the screen was blank band only the 3vg and Rx and tx (yellow and red) are on. I can't get it to reset and it won't boot into dfu mode and Ubuntu doesn't see it.

kwilkins9, 2021-09-01:
Ok I stand corrected.. it is 3 lights... The USB is so dim I couldnt see it on earlier... But I've tried to connect it to my wife's PC and my phone and still, it won't get past post or whatever. [image]

kwilkins9, 2021-09-01:
I put them on when I got the unit about 2ish weeks ago.  It worked up until tonight. I got the unit from hak5, completely bare boned. I just put on the hs from some rpi I had extra. 

I was checking out the dev stuff because I noticed that when I hit hackrf_info, it said libhack and hackrf were unknown.. so I was trying to update them but the info I had was to install all this make stuff etc.  

Anyway I was successfully running mayhem for over a week... No issues.  Only when I went to follow the steps, something went wrong.  I think at most when I tried to update the hackrf-usb.bin I am wondering if it tried to install the .bat version? But I thought the hackRF had a safety net for that with the dfu-util but I can't even get it in dfu mode

kwilkins9, 2021-09-01:
Anyway, I was also curious about all these empty spots on the hRF, so I was looking up what additional tools I could unlock by adding this or that... came to here: https://github.com/mossmann/hackrf/blob/master/firmware/README and started just reading it and installing it to practice making firmware (first time) and I got down to the toolchains and saw that all my installed hackRF tools were "unknown" and the firmware said "local ((something, I don't recall) and that's what sent me down the proverbial rabbit hole of flashing the firmware. I verified all the downloads, and reinstalled mayhem and it installed just fine... 

I wasn't in the hackRF at all when I was installing the dev  tools when the hackrf restarted and booted back up but nothing is responding.  I've taken the portapack h1 off and tried to connect it to my wife's system, my phone, my pentoo install (dual booting ubuntu/pentoo) and it's all the same.  once it is on, the reset button and dfu buttons are unresponsive.

porkiedev, 2021-09-01:
Very interesting.. You're right, the DFU is there so you can run a USB bootloader essentially making the hackrf un-brickable by software standards. Are you sure you're using the DFU correctly? I'm not familiar with it so I'm not sure how it's done.

If you can get DFU mode to work you can probably reflash the firmware and fix your problems. If you can, I would try a different version, maybe the version you used is broken.

The only other concerning thing I see is that in your image, the TX and RX lights are both on.
Sorry I couldn't be of more help, I would go search up videos or articles on how to use the DFU properly and make sure you aren't missing something, I'm not sure why the DFU wouldn't be working but if you can get it to work I would imagine that would fix your problems.

kwilkins9, 2021-09-01:
Yes I am familiar with how to use the hardware.. that is what is bothering me as well, the Rx/tx lights.. I am hoping this is a fixable issue and not just having me out $500.  I'm not getting any more money until I can find a job, so I hope it works

supersat, 2021-09-02:
FWIW, my HackRF also exhibits the same LED behavior when either it is held in reset OR it is in DFU mode. I believe it's simply an artifact of the corresponding GPIO pins on the microcontroller not being configured yet, and they default to weak pull-up inputs, causing the LEDs to glow a bit.

It seems like the firmware is corrupt or missing, and you're not getting it into DFU mode somehow. There are a handful of bootstrap pins that define the boot source -- check the voltage at test points P73 and P74 and make sure they're both 0V?

kwilkins9, 2021-09-02:
Actually I figured it out a couple of hours ago and I replicated it 3x... I fixed it by instead of restarting it with the dfu button, I used the reset button. Nothing happened but when it came back up, all but the 1ve was solid.  Then forwhatever reason, I just repeated pressed the reset about 10 times and the lights dimmed.. then I saw that it registered with hackrf_info so I flashed it again and now it works as if nothing happened.  Small victories I guess.

straithe, 2021-09-02:
Congratulations! Since you've found a solution I am going to close this issue. If you need further assistance or have another issue come up, please re-open this GitHub issue or open another one.

closed 2021-09-02 by straithe

supersat, 2021-09-02:
> ... instead of restarting it with the dfu button, I used the reset button. 

Perhaps that's not exactly what you meant, but for anyone else potentially having this issue, proper way to get it into DFU mode is to hold down the DFU button while plugging it in, OR holding the DFU button down while pressing the RESET button. The DFU button doesn't really do anything except in the few milliseconds the processor is coming out of reset.