greatscottgadgets/hackrf #943: Question about LED flashing

Noname12345GER, 2021-09-01:
### Steps to reproduce
none.

### Expected behaviour
i dunno.

### Actual behaviour
What does it mean when the 3 rightmost LEDs (usb, rx, tx) are flashing together?

### Version information
Portapack Mayhem Firmware.

porkiedev, 2021-09-01:
I went searching for answers on the Mayhem firmware github page and noticed you also submitted an issue there.

I suggest you close this issue as I believe your issue is probably resonating from the Mayhem firmware, I would redirect to your other issue submission on the Mayhem firmware page and wait for a proper answer there.

Noname12345GER, 2021-09-01:
I wanted to leave this issue open here too, because i do believe that the behavior is listed nowhere else on the hackrf documentation and therefore might happen to someone else.

Also, it might be an hackrf problem,  after all.

porkiedev, 2021-09-01:
Got it. Though, if it was a hackrf problem, I truly feel that it would've been documented somewhere. The HackRF isn't cheap to replace, so I hope you can get your issues resolved!

As I am typing this I just realized, there are 2 issues I responded to, I realized both of you guys are suffering the same issue. USB, TX, and RX lights on, the poster just replied and mentioned he was running Mayhem and it occurred after he updated the firmware and the HackRF is un-responsive. You might monitor [https://github.com/mossmann/hackrf/issues/942](url) in case they get an answer too. I'm thinking there may be an issue with the mayhem firmware.

Noname12345GER, 2021-09-01:
Thank you. I have read thru the post you mentioned, but it is of no help sadly.

porkiedev, 2021-09-02:
You might check it again. That issue was solved as the user was able to boot into DFU. I would try to get your hackrf into DFU mode and reflash the firmware and see if that resolved your issue.

Noname12345GER, 2021-09-02:
I am also able to boot into DFU,  i tried the stock firmware and the mayhem firmware. 

Both run fine on the hackrf, but still cause the flashing leds after i attach the portapack.

porkiedev, 2021-09-02:
Unusual.. I may not be of much help than as that sounds like a hardware issue. I hope your issue gets resolved soon.

Noname12345GER, 2021-09-02:
I also do believe it is an hardware issue. It has to do with the reset button. I am not familiar with the portapack hardware,  or if a seperate portapack firmware exist (unrelated to the "normal" firmware).

So something funny seems to be happening with the portapack, with causes the hackrf to "panic".

straithe, 2021-09-10:
What can we try to reproduce your problem? Can you give us a step-by-step process for what you are trying? Can you please also post pictures of your Portapack and HackRF (separately)?