greatscottgadgets/hackrf #945: gnuradio-companion install not recognized after reboot

somebodyelse87, 2021-09-05:
### Steps to reproduce
1. Linux Mint os
2. Execute pybombs install of gnuradio as per instructions

### Expected behaviour
When I type command:
gnuradio-companion

I expect the radio companion to boot.

### Actual behaviour
After the PC is rebooted, I must re enter the command
source ~/prefix-3.8/setup_env.sh
in order to start the radio companion again

once that command is entered, the gnuradio-companion boots by command as expected.  Otherwise after each PC reboot I get the message:

Command 'gnuradio-command' not found, but can be installed with:
sudo apt install gnuradio

I do not wish to install gnu-radio from the mint repositories so I cannot use apt.

### Version information
**Operating system**:
Linux Mint
**hackrf_info output:**

If you are reporting a problem that involves third party software 
(GNU Radio, Gqrx, etc), please report the version here.

### Output
```
Insert any commandline or build output here
```

straithe, 2021-09-10:
I'm going to attempt to maybe point you in the right direction here, but please note that this particular repository relates the HackRF One device and its native drivers and tools only. You might not get the help you need since it seems that your issue is not with the HackRF One, but with gnuradio-companion and Linux Mint. 

To confirm, are these the instructions that you followed?: https://wiki.gnuradio.org/index.php/InstallingGR#Using_PyBOMBS

somebodyelse87, 2021-09-11:
yes, those are the instructions.  I ended up adding the sources to the .bashrc file to make it autorun the sources command when I log in.  Unless there is a better way to do this.

straithe, 2021-09-11:
To confirm, did that solve the issue you were experiencing?