greatscottgadgets/hackrf #946: HackRF not connecting via USB

kaos713, 2021-09-05:
Hi, I tried plugging in my HackRF for the first time in a while and it's not connecting at all. Hoping someone can point me in the right direction.

### Steps to reproduce
Plug in HackRF One using USB.

### Expected behaviour
USB/RX/TX lights illuminate.

### Actual behaviour
3V3 and 1V8 indicators illuminates green. RF indicator illuminates red. USB, TX and RX indicators do not illuminate.

### Version information
**Operating system**: Ubuntu 20.04.3

I checked these issues:
https://github.com/mossmann/hackrf/issues/456
https://github.com/mossmann/hackrf/issues/569

I ran 'lsusb': HackRF is not in the list.
I ran 'hackrf_info': 'No HackRF boards found.'

I booted into DFU Mode (by holding DFU button and pressing reset), no change.
I've tried multiple USB cables, USB ports, and computers, no change.

straithe, 2021-09-10:
What OS are you using? Are you trying to use your HackRF on a bareboard machine or in a VM?

kaos713, 2021-09-10:
> What OS are you using? Are you trying to use your HackRF on a bareboard machine or in a VM?

I'm have a fresh install of Ubuntu running on a laptop, not a VM.

straithe, 2021-09-10:
Are you using the HackRF on its own, or are you using it with a Portapack?

kaos713, 2021-09-11:
> Are you using the HackRF on its own, or are you using it with a Portapack?

Yup just on its own.

straithe, 2021-09-11:
Have you tried DFU mode?

kaos713, 2021-09-11:
> Have you tried DFU mode?

Yes. I tried booting before and after plugging it in via USB but it still doesn't connect (no USB lights flash and Ubuntu reports 'unable to enumerate USB device'.

straithe, 2021-09-11:
Can you add a picture of your HackRF here?

closed 2021-10-08 by straithe

nxnjz, 2025-01-20:
Had a similar issue with an R9, solved it by plugging into a USB 2.0 port instead of a 3.0.