greatscottgadgets/hackrf #955: Leave antenna bias tee disabled while RFFC5071 is not enabled

cjheath, 2021-09-18:
https://github.com/mossmann/hackrf/blob/5f1536cc3631f79e6beafbd8d7bb6cf86dcf555f/firmware/common/rffc5071.c#L65

The initial value for the RFFC5071 GPO register should set GPO1 to logic high, leaving the bias tee disabled until required.

I think the correct value here is `((01<<12) | (01<<6) | 02)`, which sets GPO1 high in both modes, and sets the gate bit so the value is unaffected by ENABLE.

Brumi-2021, 2021-09-25:
I am José Luis. , I found accidentally that issue in the Portapack HackRF FW when connecting an external LNA into the Antenna with LED indicator of the DC bias power supply . As the LNA needed a high drive current (>180mA) I intentionally disabled the Title bar Icon of the HackRF DC-Ant-bias off , but eventually the LNA_LED was flashing at every HW USB battery cold reset . 

 It seems a common problem To the HackRF  FW . (But maybe timings of the pulse might be slightly different ) 

I investigated only in the Portapack HackRF FW , and I managed to minimise the problem a lot ,NOW  almost negligeable .
i proposed a pull request tonthe Mayhem branch with that c/m

You can find all related details in that pull request 
[https://github.com/eried/portapack-mayhem/pull/413](url)

miek, 2021-12-01:
Closing this as the proposed solution didn't work out, the issue itself is now described in #1006.

closed 2021-12-01 by miek