greatscottgadgets/hackrf #956: Upgrade to 2021.03.1 does not work

stdevPavelmc, 2021-09-19:
### Steps to reproduce
1. Follow instructions to compile host tools and firmware
2. Install compiled tools and firware
3. Try to use Gqrx, hackrf_sweep and no joy, both fail

### Expected behaviour
 Gqrx, hackrf_sweep, etc working

### Actual behaviour
None work.

Gqrx detect it do nothing once you hit play see debug output:

```
# gqrx
gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.8.1.0
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp 
gr::log :WARN: file_source0 - file size is not a multiple of item size
Resampling audio 96000 -> 48000
BandPlanFile is /home/pavel/.config/gqrx/bandplan.csv
BookmarksFile is /home/pavel/.config/gqrx/bookmarks.csv
[INFO] [UHD] linux; GNU C++ version 9.2.1 20200304; Boost_107100; UHD_3.15.0.0-2build5
gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.8.1.0
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp 
Using HackRF One with firmware git-1408473
gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.8.1.0
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp 
Using HackRF One with firmware git-1408473
```

hackrf_sweep output:

```
# hackrf_sweep -f 100:120
call hackrf_sample_rate_set(20.000 MHz)
call hackrf_baseband_filter_bandwidth_set(15.000 MHz)
Sweeping from 100 MHz to 120 MHz
Stop with Ctrl-C
0 total sweeps completed, 0.00 sweeps/second

Couldn't transfer any data for one second.

Exiting... hackrf_is_streaming() result: HACKRF_TRUE (1)
Total sweeps: 0 in 1.00176 seconds (0.00 sweeps/second)
hackrf_close() done
hackrf_exit() done
exit
```

### Version information
**Operating system**:
```
# cat /etc/os-release
NAME="Ubuntu"
VERSION="20.04.3 LTS (Focal Fossa)"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Ubuntu 20.04.3 LTS"
VERSION_ID="20.04"
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
VERSION_CODENAME=focal
UBUNTU_CODENAME=focal
```

**hackrf_info output:**
```
# hackrf_info 
hackrf_info version: git-1408473
libhackrf version: git-1408473 (0.6)
Found HackRF
Index: 0
Serial number: 0000000000000000909864c8338a5bcf
Board ID Number: 2 (HackRF One)
Firmware Version: git-1408473 (API:1.05)
Part ID Number: 0xa000cb3c 0x005e4768

```

If you are reporting a problem that involves third party software 
(GNU Radio, Gqrx, etc), please report the version here.

### Output
Gqrx see output cited above.

Gqrx was compiled locally by the master branch of the official repository, no mods.

stdevPavelmc, 2021-09-19:
Addenum: It was working just fine with the 2018 firmware, just did a delayed upgrade and it does not work now.

pkg from repository:
```
# dpkg -l | grep hackrf
ii  hackrf                                        2018.01.1-2                                amd64        Software defined radio peripheral - utilities
ii  libhackrf-dev                                 2018.01.1-2                                amd64        Software defined radio peripheral - development
ii  libhackrf0:amd64                              2018.01.1-2                                amd64        Software defined radio peripheral - runtime library
ii  soapysdr-module-hackrf:amd64                  0.3.3-3build1                              amd64        HackRF device support for SoapySDR (default version)
ii  soapysdr0.7-module-hackrf:amd64               0.3.3-3build1                              amd64        HackRF device support for SoapySDR
```

best4username, 2021-09-19:
Same as my issue https://github.com/mossmann/hackrf/issues/936

straithe, 2021-09-19:
As mentioned by @best4username, you are likely experiencing the same issue as #936. As mentioned over there, I've found that the compiler may be one of the factors at play and I suggest trying to compile the firmware again but with the compiler described here: https://github.com/mossmann/hackrf/blob/v2021.03.1/firmware/README

I am still going to do some more investigation into the HackRF code, but hopefully this will get your instance working well while I continue this investigation.