greatscottgadgets/hackrf #958: hackrf one does not react

smlyclts, 2021-09-20:
Steps to reproduce
not connecting to computer
only 3.3v led is on
not connecting to pc in dfu mode
Expected behaviour
HackRF should be detected in PC. USB LED should glow

Actual behaviour
only 3.3v led light is on

Version information
windows 10, Linux

hackrf_info output:
Unable to connect to computer, hackrf info unknown

If you are reporting a problem that involves third party software
(GNU Radio, Gqrx, etc), please report the version here.

Output
I am using clone hackrf one portapack h1 i connect power with power bank but portapack screen turned on and off

only 3.3v led light is on, i disconnected the portapack from hackrf and plugged it into the computer

this is how it shows up in device manager "unknown USB device (device descriptor request failed)"

i changed cable it because the might be damaged, i changed the power supply, same result

I suspect a short circuit but I can't detect it, I don't have a thermal camera to find the heated part, but I found the heated part by 

trying a primitive method. I froze the card using lighter gas and then plugged in the usb cable and found the part where the ice melted

[image]
[image]
[image]

smlyclts, 2021-09-20:
I removed u15 with Straithe suggestion and got no result
[image]
s

straithe, 2021-09-27:
I'm sorry but this does look like a hardware problem and this board does not look like a Great Scott Gadgets produced HackRF. I suggest contacting your seller to see if they will send you a replacement.

closed 2021-09-27 by straithe