greatscottgadgets/hackrf #959: Can't TX

Flavbf, 2021-09-21:
Hello,

i got a problem with my real (not chinese one) brand new HackRF + Portpack H1, i simply can't TX something. The led normally turn on when i'm in TX mode, but in my receptor i only got "silence". I mean, it just TX something but silence. I tried many TX mode the Mayem firmware allow to use.
I installed the Mayhem 1.4 Firmware. I also tried with the old Havoc firmware and with Mayhem 1.3 too.
My receptor is just near. I got ANT500 and 700, tried with both.

Do you think i got hardware problem or something ?

Thanks a lot.

straithe, 2021-09-21:
What happens when you remove the HackRF from the Portapack and run `hackrf_info`?

Flavbf, 2021-09-21:
Hello,

i got this when running hackrf_info with the PortaPack and Mayhem 1.3.0.

>hackrf_info version: git-7047252
libhackrf version: git-7047252 (0.6)
Found HackRF
Index: 0
Serial number: 0000000000000000f77c60dc289132c3
Board ID Number: 2 (HackRF One)
Firmware Version: local-883a622 (API:1.04)
Part ID Number: 0xa000cb3c 0x00544762
 

And this without the PortaPack (i reflashed to the HackRF original firmware, lastest version)

>hackrf_info version: git-7047252
libhackrf version: git-7047252 (0.6)
Found HackRF
Index: 0
Serial number: 0000000000000000f77c60dc289132c3
Board ID Number: 2 (HackRF One)
Firmware Version: 2021.03.1 (API:1.04)
Part ID Number: 0xa000cb3c 0x00544762

Thanks you.

straithe, 2021-09-27:
You mention a receptor. What device are you trying to receive with? Do you have a second SDR to try to receive with?