greatscottgadgets/hackrf #965: Gnu Radio does not have osmocom for HackRF

ciric50, 2021-10-01:
### Steps to reproduce
1. Installed hackrf through synaptic
2. Followed the steps at  https://wiki.gnuradio.org/index.php/InstallingGR to install gnuradio
3. Plug in HackRF and run hackrf_info. Seems fine.
4. Run gnuradio and follow the first tutorial for hackrf.

### Expected behaviour
Should be an osmocom driver to get input from the hackrf

### Actual behaviour
There is no osmocom block in gnuradio

### Version information
Linux Mint 20.1 Ulyssa
GRC version 3.9.2.0

**hackrf_info output:**
hackrf_info version: unknown
libhackrf version: unknown (0.5)
Found HackRF
Index: 0
Serial number: 0000000000000000f77c60dc2a784fc3
Board ID Number: 2 (HackRF One)
Firmware Version: 2021.03.1 (API:1.04)
Part ID Number: 0xa000cb3c 0x006a4764

```

ciric50, 2021-10-02:
UPDATE: I installed gr-osmosdr and then had a osmocom source block in GRC, but I got the following errors. Appears GRC is still not recognizing hackrf.

qt5ct: using qt5ct plugin
gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.8.1.0
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp 
[INFO] [UHD] linux; GNU C++ version 9.2.1 20200304; Boost_107100; UHD_3.15.0.0-2build5
[ERROR] [USRP2] USRP2 Network discovery error send_to: Operation not permitted
[ERROR] [X300] X300 Network discovery error send_to: Operation not permitted
[ERROR] [OCTOCLOCK] OctoClock network discovery error - send_to: Operation not permitted
[ERROR] [UHD] Device discovery error: send_to: Operation not permitted
[ERROR] [UHD] Device discovery error: send_to: Operation not permitted
[ERROR] [USRP2] USRP2 Network discovery error send_to: Operation not permitted
[ERROR] [X300] X300 Network discovery error send_to: Operation not permitted
[1m[31m[ERROR] SoapySSDPEndpoint::sendTo(udp://239.255.255.250:1900) = -1
  sendto(udp://239.255.255.250:1900) [1: Operation not permitted][0m
[ERROR] [UHD] Device discovery error: send_to: Operation not permitted
[ERROR] [UHD] Device discovery error: send_to: Operation not permitted
[INFO] Using format CF32.

straithe, 2021-10-18:
Thanks for reaching out! I need a few more details in order to help you out. 

* Did you install gr-osmosdr and GRC from source? 
* It is possible that both gnuradio 3.9.2.0 and 3.8.1.0 are installed? 
* Are there any error messages that pop up that mention HackRF?

closed 2021-11-19 by straithe