greatscottgadgets/hackrf #968: Can't install on Ubuntu

greggawatt, 2021-10-09:
I'm trying to make to update the firmware but i keep running into this error. I can't figure out how to install libopencm3 

$ make
[  1%] Performing build step for 'libopencm3_hackrf_usb'
make[3]: *** No targets specified and no makefile found.  Stop.
make[2]: *** [CMakeFiles/libopencm3_hackrf_usb.dir/build.make:112: libopencm3_hackrf_usb-prefix/src/libopencm3_hackrf_usb-stamp/libopencm3_hackrf_usb-build] Error 2
make[1]: *** [CMakeFiles/Makefile2:127: CMakeFiles/libopencm3_hackrf_usb.dir/all] Error 2
make: *** [Makefile:84: all] Error 2

miek, 2021-10-10:
libopencm3 is included as a submodule so it doesn't need to be installed separately, but you have to clone it by running:

    git submodule update --recursive --init

straithe, 2021-11-10:
I'm going to close this as there hasn't been a response in a while, but please re-open this issue or open a new one if you still need assistance.

closed 2021-11-10 by straithe

greggawatt, 2021-11-14:
Sorry I didn't update I was able to get it working, thanks for the help!

straithe, 2021-11-14:
I'm happy to hear you got some success! Thank you for the update.

noodleboy91, 2022-10-26:
> 

Just want to say thank you for this, it fixed my "*** No targets specified and no makefile found. " issue :)