greatscottgadgets/hackrf #969: How to use rad1o firmware

polygon, 2021-10-10:
Trying to use the rad1o firmware from the 2021.03.1 release package. I start the rad1o badge in MSC mode and copy `firmware-bin/hackrf_rad1o_usb.bin`, renaming the extension to `b1n`. When I start the firmware, the device won't register on USB. Anything I have to do different?

### Steps to reproduce
1. Obtain `hackrf_rad1o.bin` from 2021.03.1 release
2. Copy to rad1o in mass-storage mode, rename extension to `b1n`
3. Start the new l0dable

### Expected behaviour
Rad1o shows up on USB

### Actual behaviour
Is not seen on USB

### Version information
**Operating system**: Linux

### Output

schneider42, 2021-10-11:
I'll can probably have a look during this week. Did you already perform this successfully with an earlier release?

polygon, 2021-10-11:
I did update the rad1o in the past with the releases from https://rad1o.badge.events.ccc.de/firmware:releases and also self-built firmware from the rad1o repos but never from a hackrf release. If I've read correctly, 2021.03.1 was the first and so far only release that had rad1o support.

closed 2021-10-13 by polygon