greatscottgadgets/hackrf #978: -p flag not recognized by hackrf_transfer

Anjum9694, 2021-10-30:
### Steps to reproduce
1. hackrf_transfer -p 1

### Expected behaviour
Bias-tee being enabled

### Version information
**Operating system**: Windows 10

**hackrf_info output:**
C:\Users\Anjum\Desktop\HackRF Tools>hackrf_info
hackrf_info version: unknown
libhackrf version: unknown (0.6)
Found HackRF
Index: 0
Serial number: 0000000000000000f77c60dc277a3fc3
Board ID Number: 2 (HackRF One)
Firmware Version: 2021.03.1 (API:1.04)
Part ID Number: 0xa000cb3c 0x0066435d

### Output
```
C:\Users\Anjum\Desktop\HackRF Tools>hackrf_transfer -p 1
specify one of: -t, -c, -r, -w
Usage:
        -h # this help
        [-d serial_number] # Serial number of desired HackRF.
        -r <filename> # Receive data into file (use '-' for stdout).
        -t <filename> # Transmit data from file (use '-' for stdin).
        -w # Receive data into file with WAV header and automatic name.
           # This is for SDR# compatibility and may not work with other software.
        [-f freq_hz] # Frequency in Hz [0MHz to 7250MHz].
        [-i if_freq_hz] # Intermediate Frequency (IF) in Hz [2150MHz to 2750MHz].
        [-o lo_freq_hz] # Front-end Local Oscillator (LO) frequency in Hz [84MHz to 5400MHz].
        [-m image_reject] # Image rejection filter selection, 0=bypass, 1=low pass, 2=high pass.
        [-a amp_enable] # RX/TX RF amplifier 1=Enable, 0=Disable.
        [-p antenna_enable] # Antenna port power, 1=Enable, 0=Disable.
        [-l gain_db] # RX LNA (IF) gain, 0-40dB, 8dB steps
        [-g gain_db] # RX VGA (baseband) gain, 0-62dB, 2dB steps
        [-x gain_db] # TX VGA (IF) gain, 0-47dB, 1dB steps
        [-s sample_rate_hz] # Sample rate in Hz (2-20MHz, default 10MHz).
        [-n num_samples] # Number of samples to transfer (default is unlimited).
        [-S buf_size] # Enable receive streaming with buffer size buf_size.
        [-c amplitude] # CW signal source mode, amplitude 0-127 (DC value to DAC).
        [-R] # Repeat TX mode (default is off)
        [-b baseband_filter_bw_hz] # Set baseband filter bandwidth in Hz.
        Possible values: 1.75/2.5/3.5/5/5.5/6/7/8/9/10/12/14/15/20/24/28MHz, default <= 0.75 * sample_rate_hz.
        [-C ppm] # Set Internal crystal clock error in ppm.
        [-H hw_sync_enable] # Synchronise USB transfer using GPIO pins.
```

straithe, 2021-11-03:
Have you tried combining `hackrf_transfer -p 1` with one of the suggested flags (`-t, -c, -r, -w`) dictated in the text right after you tried to run the command?

Anjum9694, 2021-11-04:
I can't use any of the "-t, -c, -r, -w" options it suggests as that starts a receive or transmit through the command line, which means I CAN'T start SDR# because the hackrf is running already in the command line.

grvvy, 2021-11-04:
Could use a bit more context here. What exactly are you trying to do?

Anjum9694, 2021-11-05:
Expected behaviour:

Bias-tee being enabled

straithe, 2021-11-05:
After you enable bias-tee, what are you hoping to do? Can you give more context?

Anjum9694, 2021-11-05:
I want to use Hack-RF bias-T to power my LNA

straithe, 2021-11-05:
The `-p` flag only works while streaming data to or from the HackRF at this time. You can check if SDR# provides an option for you to enable bias tee.

straithe, 2021-11-05:
You could try to do a short transfer and enable the bias tee with HackRF. The bias tee should stay on after the transfer.

Anjum9694, 2021-11-10:
> The `-p` flag only works while streaming data to or from the HackRF at this time. You can check if SDR# provides an option for you to enable bias tee.

SDR# doesn't have this option. I tried with another software with Bias-T and it works! So I don't know any way to use SDR# hackrf with bias-T.

> You could try to do a short transfer and enable the bias tee with HackRF. The bias tee should stay on after the transfer.

It works BUT, as soon as I close the cmd window the RX button goes out along with bias-T power.

I read this in the HackRF documentation:
```
The DFU button only invokes the bootloader during reset. This means that it can be used for other functions by custom
firmware.
```

So is it possible to program the DFU button for enabling/disabling Bias-T?

straithe, 2021-11-10:
I'm happy to hear that another software option works for you! What software did you end up using? 

If you write your own firmware you could possibly repurpose that button.

closed 2021-11-10 by Anjum9694

straithe, 2021-11-10:
Check out this documentation: https://github.com/greatscottgadgets/hackrf/blob/master/firmware/README