greatscottgadgets/hackrf #980: HackRF One Transmission Tones

watsonlg, 2021-11-01:
Hello,

I was interfacing with the HackRF via command line input. When transmitting at high frequencies, for example 6 GHz, a spectrum analyzer showed the two tones used to generate the 6 GHz frequency. I believe these are the result of the IF and LO stages.

1. It seems as the result of the architecture, the HackRF One is not capable of a creating a pure tone 6 GHz signal. Is this correct?
2. Does the HackRF One have any internal filtering on the Tx end to attenuate the tones used to generate the high frequency signal?

Thank you!

straithe, 2021-11-05:
Yes, you are correct. The HackRF One is not capable of creating a pure 6 GHz signal. 

HackRF does have some internal filtering but maybe not as much as you are looking for. You may need to add external filtering for some applications.

watsonlg, 2021-11-05:
Thank you straithe, 

Do you know if the internal filtering is enabled by default or does the filter need to be enabled?

straithe, 2021-11-08:
Internal filtering is always enabled.

straithe, 2021-11-11:
Do you have any other related questions, or may I close this issue now?

closed 2021-12-14 by straithe