greatscottgadgets/hackrf #983: Ship the udev rules upstream

AsciiWolf, 2021-11-05:
Please, consider submitting the [udev rules](https://github.com/greatscottgadgets/hackrf/blob/master/host/libhackrf/53-hackrf.rules) to upstream (systemd). Here is an example how to do it: https://github.com/systemd/systemd/pull/20561

Without this, it is [not possible](https://github.com/flatpak/xdg-desktop-portal/issues/227#issuecomment-946760482) to use HackRF SDR devices in Flatpak and other special configurations where it is not possible to install the HackRF libs/udev rules system-wide.

straithe, 2021-11-08:
Thank you for the suggestion. I think it is a great idea. I will try to get to this as soon as I can, but my schedule is full for tasks until at least December.

straithe, 2022-09-27:
After deliberating our team has decided not to submit the udev rules upstream to systemd. Thanks.

closed 2022-09-27 by straithe

AsciiWolf, 2025-04-21:
https://github.com/systemd/systemd/issues/37019

AsciiWolf, 2025-05-13:
If you ever re-consider this, bladeRF was already added into systemd hwdb and can be used as an example of how the changes should look like:

https://github.com/systemd/systemd/blob/main/hwdb.d/70-software-radio.hwdb

AsciiWolf, 2025-09-09:
@straithe Are we allowed to add these rules to systemd or do you want them to stay downstream? Thanks!

AsciiWolf, 2025-09-09:
Note that no action will be required from you or your team. We will add the rules and handle any possible problems.

The only thing we ask of you is to either allow or deny us doing it. :-)