greatscottgadgets/hackrf #984: hackrf_open() failed: Access denied (insufficient permissions) (-1000)

medrockstar, 2021-11-08:
Hello,

When I type hackrf_info command, I got this error in Macos Big Sur :

hackrf_open() failed: Access denied (insufficient permissions) (-1000)

Tks,

straithe, 2021-11-08:
I believe this issue can sometimes come up when HackRF is already in use elsewhere. Can you try to run the command again and confirm that you are not using any other SDR tools or software that uses HackRF at the same time

Anjum9694, 2021-11-10:
This problem is addressed in the HackRF documentation.
[HackRF Documentation.PDF](https://hackrf.readthedocs.io/_/downloads/en/latest/pdf/)

medrockstar, 2021-11-10:
Tks @straithe this is the problem, I used my Hackrf in VM Machine.

closed 2021-11-10 by medrockstar

markl17, 2023-02-12:
why am I getting this error
[INFO] Opening HackRF One #0 10961dc2975534f...
gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.10.4.0
built-in source types: file rtl rtl_tcp uhd hackrf bladerf airspy airspyhf soapy redpitaya
[INFO] [UHD] Win32; Microsoft Visual C++ version 14.2; Boost_107800; UHD_4.3.0.0-release
Traceback (most recent call last):
  File "C:\Users\markl\Downloads\top_block.py", line 231, in <module>
    main()
  File "C:\Users\markl\Downloads\top_block.py", line 209, in main
    tb = top_block_cls()
         ^^^^^^^^^^^^^^^
  File "C:\Users\markl\Downloads\top_block.py", line 108, in __init__
    self.rtlsdr_source_0 = osmosdr.source(
                           ^^^^^^^^^^^^^^^
RuntimeError: Failed to open HackRF device (-1000) Access denied (insufficient permissions)

straithe, 2023-02-12:
Please open a new issue.