greatscottgadgets/hackrf #989: HackRF commands

jerrydsp, 2021-11-10:
Please forgive me if I have posted these questions in the wrong place.

I am considering getting a HackRF SDR to use in my radio astronomy work so this is part of my research before I actually buy one of these devices. 

I would like to use the HackRF to acquire signal data and store the data in a file for off line processing. To this end I need to know how to tune this SDR, set the filter bandwidth and sampling rate and acquire the data as a file.

1. Can any body tell me where I can find a list of all the commands one may use to control and use this SDR ?
2. The I & Q channels are 8 bits wide. In the future will this be expanded to 10 or 12 bits ?

Many thanks in advance,

Jeremy.

gozu42, 2021-11-10:
> I would like to use the HackRF to acquire signal data and store the data in a file for off line processing. To this end I need to know how to tune this SDR, set the filter bandwidth and sampling rate and acquire the data as a file.

check the options for hackrf_transfer

>     1. Can any body tell me where I can find a list of all the commands one may use to control and use this SDR ?

install the hackrf host tools, type hackrf_<tab><tab>, then look at the usage info like "hackrf_transfer -h".
this does not require an actual hackrf to be connected.

>     2. The I & Q channels are 8 bits wide. In the future will this be expanded to 10 or 12 bits ?

for receiving, this is not possible with the current hardware.
for transmit, it would be possible to change to 10bit, and since the "pins" are already connected in the right places would be a software-only change, but i am not aware of anyone working in that direction.

jerrydsp, 2021-11-10:
gozu42  Thank you for your prompt reply. Much appreciated.
I will proceed as you suggest !!

About the width of the I & Q channels ... OK.
I guess one can live with a 48db dynamic range. It means I will need a better front end amplifier.

Regards,

Jeremy

straithe, 2021-11-10:
Thank you @gozu42 for the information! @jerrydsp, it looks like gozu answered your questions. May I close this issue?

closed 2021-11-10 by jerrydsp